mist-devel/mist-firmware #1: ARM -> FPGA file transfers

sorgelig, 2015-12-07:
I'm trying to understand how ARM -> FPGA file transfer works. To be precise, i want to know if i can receive file over SPI at the rate i want. Currently, in ZX Spectrum i see SPI data goes to RAM buffer directly at the rate SPI pushes it. But i wonder if it possible to receive it without RAM buffer? For example i receive one block (512b), then put reception on hold till i finish to process it, then receive another 512b and so on. Is it possible? And how?

harbaum, 2015-12-08:
Have a look at the sd card implementation of e.g. the zx spectrum core (fpga side is in sd_card.v and user_io.v and the arm controller side is in user_io.c). This reads 512 byte blocks of data from the arm controller in exactly the same way you want. Data is stored in a buffer in sd_card.v.

The Plus Too macintosh core uses the same interface directly (not through the sd card emulation layer on fpga side) to implement a scsi disk.

closed 2015-12-08 by harbaum

sorgelig, 2015-12-08:
What i mean is receiving files chosen from OSD menu. Not SD card access interface.

sleary78, 2015-12-08:
I dont think raising bugs/issues is the appropriate forum for getting support for something like this.

harbaum, 2015-12-09:
Indeed. But to answer this question, anyway: There's no method atm to throttle a file transmission triggered by the OSD.