mist-devel/mist-firmware #4: How to know the size of mounted image (S option)?

sorgelig, 2016-06-09:
If i use S option in OSD, core gets event about mounting - this is good. But i cannot see how i can get the size of mounted image. How to get it? Cannot find it in firmware sources.

harbaum, 2016-06-09:
Hi,

there's currently no way to tell how big a mounted image is. So far it was 
always possible to read this from inside the image. E.g. if the image was a 
file system the core would read the partition table and the boot sectors.

Regards,
  Till

sorgelig, 2016-06-09:
Ok. In the core I'm working on, there is no way to know the image size in advance, but i have to scan the whole image before use. I've added new command definition into FW sending the size of image.
I will test it, and if it will go smooth without affecting older cores, then i will push the commit.

harbaum, 2016-06-09:
Sounds good and will be useful for e.g. the PlusToo core which may use this when formatting hdd images.

sorgelig, 2016-06-10:
I've pushed commits and tested on SAM Coupe core. Works without problems.
See https://github.com/sorgelig/SAMCoupe_MIST/blob/master/mist_io.v for handling 'h1d command.

closed 2016-06-10 by sorgelig