mist-devel/mist-firmware #6: OSD issues after recent updates

sorgelig, 2016-06-10:
1) Joy1 has some glitch in display status (at least when no joysticks connected).
2) Keyboard test shows Amiga translated key codes (usb2ami) instead of PS2.
3) Shouldn't be more natural to press ESC key from new pages (like in other pages)?

Newsdee, 2016-06-11:
Fixed (1) and (2).  

For (3) it's because we map ESC to a joypad button by default (to allow navigation),
and for keyboard test we want to minimize the keys that close it (to display keycodes).
I'll change it for the USB status though.

Newsdee, 2016-06-11:
Improved (3) by cleaning up the OSD navigation options, now it should flow more nicely.

sorgelig, 2016-06-11:
2) still not correct. You display USB scan codes which is not correct either. Cores get PS2 scan codes. You need to use usb2ps2 translation. Please note about extended code flag (E0) as well.
3) Can you simply check both SPACE and ESC keys at the same time?

Newsdee, 2016-06-11:
2) We need USB for ease of mapping too. I've added both (USB and PS/2) which probably is best.
I'm not handling the extended code flag yet... I'd like to add that and the modifiers (shift, alt, etc).
3) That should work for the keyboard test, yes

sorgelig, 2016-06-11:
shift/alt/ctrl are not relevant in keyboard test. They are the same keys as others. But E0 code is.
Just display the code as inverted if it includes extended code.

Newsdee, 2016-06-12:
Added display of E0 (as "e" prepended) to PS2 scancodes to the C code; give it a try.
I didn't invert the display for it as I haven't found a compatible function for that. But this should do what you are after.

sorgelig, 2016-06-13:
look ok, though may be wider spacing between PS2 codes will look better.

closed 2016-06-13 by sorgelig