mist-devel/mist-firmware #7: Keyboard test OSD is broken?

sorgelig, 2016-07-13:
I've compiled firmware several days ago, and just now i've discovered keyboard test screen is broken.

Newsdee, 2016-07-13:
I've had to disable the PS/2 handling. For some reason it causes conflicts with the Amiga core... (prevents loading config) so I gave priority to get the core working again. 

I will try it again when I get a bit of time.

sorgelig, 2016-07-14:
Input test OSD is still accessible, but inside is total mess.

Newsdee, 2016-07-16:
Just checked and I don't have uncommitted code. Which core are you using? And I guess you are booting with the menu core? (I'll try to replicate it)

sorgelig, 2016-07-16:
I'm talking about this issue:
https://postimg.org/image/507z5271t/
look at that squares. Also pressed keys not displayed. You need to press 3 or 4 keys at the same time to see some changes.
I was compiled firmware from current firmware sources, not the last released.

sorgelig, 2016-07-16:
Ah, i see, squares are just modifiers (don't know why you need to display it this way since all modifiers have its key codes). 
But there is still bug when no keys displayed. You need to press 3 keys at the same time to see some display.

Newsdee, 2016-07-20:
Try the latest change in the repo, I think I've fixed the ps/2 display functionality without breaking the Amiga core.   Single keys seem to work OK but in my Rapoo keyboard I noticed it can glitch if I press certain keys together (and show none are pressed). Still should be functional for basic tests.

Also: USB HID does not have scancodes for modifiers (unlike PS/2). I do some translation to convert the HID byte into the correct PS/2 modifiers to display. See here for some description: https://docs.mbed.com/docs/ble-hid/en/latest/api/md_doc_HID.html

sorgelig, 2016-07-20:
hmm.. not fixed.
It start to display codes only if i press tree or more keys at the same time. Except modifier keys. Modifiers codes start to display from first pressed key.
Another strange thing: if i press modifiers they start to appear from the left. If i press normal keys, they start to appear from the right (if 3 or more keys pressed.)

UPD: The above is from Logitech K400r wireless keyboard.
If i connect simple wired keyboard then everything shown correctly.

Both wireless and wired keyboard work OK with cores and OSD.

sorgelig, 2016-07-20:
I've pushed the fix.
Somehow wireless keyboard fill the buffer starting from the other end.
May be other keyboards are affected as well.

Newsdee, 2016-07-21:
Ah I just saw your last reply... yes I was thinking the HID report could have been in the wrong order, and I had reduced the display to 4 keys to save some space.  We could restore 6 key display but I feel it looks a bit cleaner this way.

closed 2016-07-21 by sorgelig