mist-devel/mist-firmware #8: itoa undefined identifier?

sleary78, 2016-09-11:
I'm sure I must be doing something wrong but I tried to build the firmware using the gcc built by the auto script and also with the stock arm-none-eabi-gcc in debian (newlib installed too). 

I get the following errors (all of which seem legitimate). 

state.c: In function 'StateReset':
state.c:352:2: error: 'for' loop initial declarations are only allowed in C99 mode
  for(int i=0; i<6; i++) {

This is trivial to fix but it is a bug. After fixing this I get ...

menu.c:(.text+0x930): undefined reference to `itoa'

My code is a clean checkout with clean tools.

harbaum, 2016-09-11:
No such problems here. What gcc do you use? Mine is:

$ /opt/arm-none-eabi/bin/arm-none-eabi-gcc --version
arm-none-eabi-gcc (GCC) 5.2.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

sleary78, 2016-09-11:
arm-none-eabi-gcc --version
arm-none-eabi-gcc (4.8.4-1+11-1) 4.8.4 20141219 (release)

The install script installed 4.8.1 for me.

sleary78, 2016-09-11:
looks like i have an older version of the install script. I'd recommend explicitly stating C99 and 5.1+ gcc if thats needed.

closed 2016-09-11 by sleary78

harbaum, 2016-09-25:
These few code sections should imho rather be changed not to require c99 and the latest gcc.

sleary78, 2016-09-25:
Realistically are we going to compile this with a pre C99 compiler? C99 > 15 years old now. Even GCC has embraced C99 as default from 5.0+.

rkrajnc, 2016-09-26:
Maybe just adding -std=c99 (or gnu99) to makefile will fix this?

sleary78, 2016-09-26:
That would solve the issue.