mist-devel/mist-firmware #9: user_io core (C-64 joystick) not detected after ConfigureFpga() on xilinx artix-7

emard, 2016-10-16:
HI

I have a board ESA11 with AT91SAM7S256D and xilinx artix-7 xc7a102t and
latest mist-firmware.

After bitstream is successfuly loaded from SD card into FPGA, the
user_io.c will not detect the user_io.v core. 
Workaround: if I press reset to ARM AT91 then it skips loading bitstream 
and user_io.v core is normally detected and joystick works.

I traced problem to this:

Normally user_io.v core returns core type 0xA4 what is 8-bit joystick on DB9.
but after ConfigureFpga() does its upload, the core_io.c receives wrong core type
0x49 instead of 0xA4

I suspect the SPI interface of ARM is left in some non-default state.
I tried to force re-initialize SPI at user_io_detect_core_type() and 
retry many times using this code snipped, but it doesn't help (only ARM reset helps)

void user_io_detect_core_type() {
  core_name[0] = 0;
  int retry = 0;

  spi_init();
  EnableCard();
  DisableCard();
  EnableFpga();
  DisableFpga();
  EnableOsd();
  DisableOsd();
  EnableDMode();
  DisableDMode();

  core_type = CORE_TYPE_UNKNOWN;
  while(retry++ < 50 && core_type == CORE_TYPE_UNKNOWN)
  {
    TIMER_wait(1);
    EnableIO();
    TIMER_wait(1);
    core_type = SPI(0xff);
    TIMER_wait(1);
    DisableIO();
    iprintf("Detecting core type (%02x), retry %d    \n", core_type, retry);

    if((core_type != CORE_TYPE_DUMB) &&
     (core_type != CORE_TYPE_MINIMIG) &&
     (core_type != CORE_TYPE_MINIMIG2) &&
     (core_type != CORE_TYPE_PACE) &&
     (core_type != CORE_TYPE_MIST) &&
     (core_type != CORE_TYPE_ARCHIE) &&
     (core_type != CORE_TYPE_8BIT))
    core_type = CORE_TYPE_UNKNOWN;
  }

harbaum, 2016-10-17:
The Xilinx code hasn't been used for years now. The arm you use is the same as the one on the MIST.

But i don't see the relationship between the bitstream upload and the spi interface. The bitstream upload uses bit banging and should not affect the SPI interface at all.

emard, 2016-10-17:
HI

I see, not much testing for xilinx recently

Maybe it is possible that this bitbang xilinx SPI interface is also shared with
SPI and/or chipselect that is used for user_io.v communication but after
the bitbang finishes it is left in different state that affects user_io.v

for example something in file
https://github.com/mist-devel/mist-firmware/blob/master/fpga.c
line 213

    // disable outputs
    *AT91C_PIOA_ODR = XILINX_CCLK | XILINX_DIN | XILINX_PROG_B;

Or enabling DMA transfer for SD card traffic and not disabling it properly
after bitstream upload (user_io.c I guess doesn't use DMA to send
joystick activity)

harbaum, 2016-10-17:
"recently" ? The Xilinx code in the MIST firmware has never been used. It's still there from the original Minimig. It's there for people like you who want to re-activate the Xilinx part on some other hardware. But i have never ever used it and i am surprised that it works at all ...

emard, 2016-10-17:
:) you can consider it again in use :)

I think I found the problem and fixed it

First the -O2 is a bad idea for the gcc,
-O1 must be used, second yes it was
a line I mentioned before had to be changed
a little

I will repeat it on your fresh cloned tree
with minimal changes which fix it
and make a pull request

emard, 2016-10-18:
Thanx for the merge, seems it is fixed now

closed 2016-10-18 by emard