mist-devel/mist-firmware #12: SD card access is extremely slow on latest FW

sorgelig, 2016-12-22:
Just compiled and flashed latest FW and found out it has very slow SD card speed. Originally i've thought it's because of SD card extension cable (i'm adding MiST into another case). Then i've flashed FW from my last commit and got much faster SD speed.
I'm not 100% sure what commits affected the performance, but i guess those SPI related from Emard.

sorgelig, 2016-12-22:
what those SPI commits are for?

sorgelig, 2016-12-22:
ok. I've read closed issues and found out it's for Xilinx.
I suggest to revert these commits and ask Emard to make a conditional compilation for his board and re-submit new version of his commits.

sorgelig, 2016-12-22:
this commit: https://github.com/mist-devel/mist-firmware/commit/4c4abb62fd8bc38e41c029581f93c743f9a00c1a
is the source or problem

harbaum, 2016-12-22:
I'll have a look asap. And yes, if they have negative side effects they should be removed.

commit 602942e9 (2016-12-22): Undo patch esa11: user_io.c fix -> re-intialize SPI, make sure all other, fix #12
closed 2016-12-22 by harbaum