mist-devel/mist-firmware #21: Incorrect controller button mapping

hippi-viking, 2018-11-20:
As per [lazyweb/wiki/GameControllerUSBDatabase](https://github.com/superjamie/lazyweb/wiki/Game-Controller-USB-Power-Database) and [OpenEmu/issues/792](https://github.com/OpenEmu/OpenEmu/issues/792) it seems that several different controllers are built using the same board identified by vid:0x0079, pid:0x0011. (Mine is probably [this](https://www.amazon.com/Retro-Bit-Retrolink-Classic-Controller-Compatible-/dp/B002BFVAV0?SubscriptionId=AKIAILSHYYTFIVPWUY6Q&tag=duckduckgo-d-20&linkCode=xm2&camp=2025&creative=165953&creativeASIN=B002BFVAV0) Genesis one with the same vid:0x0079, pid:0x0011 - recognised as Retrolink NES controller by Mist.)
The problem is that the buttons are not recognised as they should be (they seem to work under Windows though).

This is how Windows and Mist recognises the button presses:
`Windows || 1 | 2 | 3 | 4 | 5 | 6 | 7 | 8 | 9 |`
`M i s t || 2 | 1 | 2 | - | 5 | 6 | 7 | 8 | 9+3 |`

Legend (as per Windows buttonmap):
`..7.........8..`
`D-pad.....4 5 6`
`D-pad..9..1 2 3`

note 1: Hope this is clear, I couldn't yet figure out a better way to show all this in text.
note 2: The recognised button presses are (of course) independent from the buttonmap one can define in the .ini file.

Quickly reading through the relevant parts of the Mist firmware source code I could not find any hardcoded custom mappings for the Retrolink NES controller which might be the culprit for this. Could this be a bug of the button readout/translate function?

Thank you for all the hard work on the code (especially for the recent multi-button controller patch)! :)

gyurco, 2018-12-18:
Maybe this helps?
http://atari-forum.com/viewtopic.php?f=115&t=28174&start=75#p359712

hippi-viking, 2018-12-19:
Wow, thanks! This seems very promising.

hippi-viking, 2019-01-12:
working fine now, thanks for accepting my patch!

closed 2019-01-12 by hippi-viking