mist-devel/mist-firmware #31: Core names and file names

jotego, 2019-07-14:
I am writting now the core for the **Commando** arcade. Following the name convention for all my cores, I am calling it **JTCOMMANDO**. However, the firmware does not send the JTCOMMANDO.rom file automatically to the core. Is there a file name length limitation?

gyurco, 2019-07-14:
8+3, as in DOS.
Don't know if LFN support can be added for config/rom files, there's a lot of places when the 8+3 naming convention is used.
Luckily plain COMMANDO is exactly 8 chars length :)

closed 2020-01-12 by jotego

jotego, 2024-04-13:
Is this still the case? Can core names be longer than 8+3 in modern firmware?

gyurco, 2024-04-13:
No, there's no practical limit now. The RBF= in the ARC file is limited to 16 chars, but it's easy to increase and also not an issue, I think.