mist-devel/mist-firmware #37: Proposal to remove controller swap from the firmware

gyurco, 2019-11-08:
The controller swap is only because of two cores: ST and Amiga. However there are 50+ cores, which don't need this. Now if only one USB controller is connected, you have to use the swap joysticks option to make it work. The mouse is handled separately, so no point to renumber the controllers because of that. DB9 mice also not supported. I think it would be easier to swap the controllers in those two affected cores than to force everyone to use "swap controllers".
My proposal is to make the controller order straightforward:
0...n-1 => USB controllers (skip when n=0)
n,n+1 => DB9
if prefer DB9 in mist.ini set:
0 - DB9 port 0
1..n => USB controllers (skip when n=0)
n+1 - DB9 port 1

In ST and Amiga, port 1 will get joy0, port 0 will get joy1 automatically.
Opinions?

squidrpi, 2019-11-08:
Can you explain in more detail what is happening here? So you're saying with the Megadrive core, as an example, the USB controller setting needs to be set to swap because of the firmware swap option?

Also C64 needs the joystick swap option for some games but in future this could be implemented in just the core? For the C64 it's nice not having to physically swap the ports. Can't think of any other cores at the moment that require a swap, perhaps Atari 800, but can't remember.

gyurco, 2019-11-08:
The firmware currently returns controller 1 as controller 2 (or 0 as 1), and controller 2 as controller 1 (1 as 0) to the cores, because on Amiga and ST you plug the first joystick into the second port (the first port is for the mouse). However this is confusing for other cores (you have to use the swap joystick option in every core, right? - if there's no such option, like in lot of arcade cores, then it just works, because it equally watches the 2nd controller - however if there's a 2 player option, you're doomed). My proposal is to make this swap only at the core level for Amiga and ST, on the rest you can decide about the exchange, if there's a swap joystick option in OSD. On the user level most probably you'll only notice that you have to turn off "swap joystick".

harbaum, 2019-11-08:
In theory i'd instantly say yes. But this would break so many existing cores or at least would change their behaviour. The same effect would be if we just rename joystick1 and joystick2 in userio, wouldn't it? After all we always transfer all joysticks over spi just not in the obvious order.

gyurco, 2019-11-08:
Checked some cores where the controllers are swapped in the code, and no Swap option in the OSD:
_Ordinary cores:_
Archie
NES
Atari800
_Jotego's cores:_
All CAPCOM arcades (common controller code - change needed only at 1 place)
_Gehstock's arcades:_
Moon Patrol

Not much work I think (+ST, Amiga).
I think swapping user_io now is a bit late.
Maybe the first step would be reverting these to the "normal" order, and adding a swap option to the OSD, then the firmware could be changed. Meanwhile if I found other cores which needs a change, I'll add to this list.

harbaum, 2019-11-12:
Ok, then I can live with that :-)

alessioscand, 2023-11-01:
Hello,
I'd like to ask, please, to take Gyurco's far-sighted proposal back into consideration.

I understand the "historical" reason of course, but today IMHO the default joystick swap by firmware is more inconvenient than useful.

In 90% of the cores, a User must also enable, in addition, the Joystick Swap option of each core for seemingly "no reason", and if such option is not available (as in many arcade implementations) the core becomes uselessly cumbersome. 

I'll give a practical example with any episode of Street Fighter II (CPS core that doesn't offer a Joystick Swap option).

If I insert just one coin and press Start, nothing happens. If I insert more than one coin and press Start, I'm Player 2: the character selection screen appears with both P2 (controlled by me) and P1 (controlled by... nobody) and I have to wait for time to "run out" to start a game against... a "ghost" P1. At the end of two rounds against this immobile player, the single player game finally begins but, being P2, my character is always on the right side, instead of the canonical left.
The arcade game assumes that if you press Start on Player 2 there must be also a Player 1, and there is no way to change this behavior, nor via Service menu or dipswitches. The CPS core does not offer a Joystick Swap option, the User is doomed.
This happens for many other cores, and I can feel the confusion of a User who ignores this "default swap" applied by the firmware (like me, he will think that that the mapping is not correct, that the joystick does not work properly, that it depends on the USB port etc.).

IMHO some kind of option in _mist.cfg_ that would allow to disable this firmware behavior would be more than welcome in certain contexts.

Thank you so much.

lroby74, 2024-04-21:
Hello, 
is possible to add a swap joystick option in OSD minimig core menú?
I have a Miss 500 and a Sidi128 where there are some issues on actual configuration.
On Miss 500 joystick and mouse are connected swapped between them respect real Amiga 500 and on Sidi128, only 1 port has +5V and Autofire function and is the port for player 2.
This is very frustrating, thank you