mist-devel/mist-firmware #39: Direct SD->FPGA ROM upload

gyurco, 2019-11-25:
I want to add direct SD-> FPGA upload for ROMs. This requires a small change in the user_io_file_tx_send() function (and implement SS4 in the cores). However I don't want to break (nor convert) all existing cores. Somehow the core should indicate it accepts direct transfers. I think about using one bit as a flag in the core type, but its bitmask currently 0xa7, which filters all the implemented types. Probably 0xaf would be better for future-proofing.
There are 2 bits remaining, however using these would be very ugly. Or adding a 8BIT2 type for just this feature?
Any better idea?

harbaum, 2019-11-25:
Sounds good to me. The A7 mask was there to prevent problems with unreliable spi transfers. This is probably one of the first code lines that ever existed in that firmware and at that time there were all kinds of problems which i tried to fix to make the whole thing even boot ....

gyurco, 2019-11-25:
The mask is no problem, but using e.g. bit6 (or bit4) to indicate direct transfer is supported should be OK? Hopefully no glitch will send x"E" instead of x"A" as the first nibble of the core type.

harbaum, 2019-11-25:
I don't expect problems. I'd use bit 4 as you then have a 1 and 0 before that may break in case of transfer problems.

gyurco, 2019-11-25:
Ok, then let's use bit4!

gyurco, 2019-11-25:
Titan 2 demo download speed (8MB): 25 sec now, 13 with direct download.

harbaum, 2019-11-25:
Wow. Sounds really cool.

gyurco, 2019-11-25:
Pushed the firmware change and the fpgagen change.
But it's still just ~5MBit/s. The SPI clock is 24 MHz. I wonder if it can be speeded up more.

gyurco, 2019-11-27:
@harbaum Do you remember why do you marked MMC_ReadMultiple with !!!!!?
https://github.com/mist-devel/mist-firmware/blob/398fb6abb0628270da9dfee6aba8c3c82224be7d/fat.c#L1216
Minimig HDD uses this, so probably not buggy. And it speeds up the transfer by another factor of 2 (at least Miniming HDD benchmarks shows 2x speed than ACSI on ST).

closed 2019-12-05 by gyurco