mist-devel/mist-firmware #41: MRA support

jotego, 2020-01-12:
As you probably know, MiSTer is now supporting an XML file called MRA which contains the details about how to transfers files from the SD card to the core. This removes the need of creating batch files to generate ROM files from several individual files of a MAME rom set.
It would be nice if MiST could support this format too.
An alternative approach would be to write a PC executable that generates the MiST ROM file using the MRA, so at least all cores can use the same tool.

harbaum, 2020-01-12:
How does that work? Are the roms loaded in one big continuous chunk? The MIST supports that since a long time using the ini file. Did you ever try that?

harbaum, 2020-01-12:
See https://github.com/mist-devel/mist-board/wiki/DocIni#rom

jotego, 2020-01-12:
The ini file approach is quite similar. But it only allows a ROM per core (as far as I know). But for instance, the Double Dragon core can load 6 different rom sets. If the OSD F option displayed also entries in the ini that could work too. But it should support more than one entry per game, and each entry many files to be sent one after the other.

MRA files are going to support DIP settings too. This takes the DIP definition from the core files to an external file. I think this mostly serves people who want to add support for game alternative ROMs which the creator didn't use.

MRA files also support selecting the file to load from its CRC sum check. This prevents the problem of files with different names, which is quite common in ROM files.

gyurco, 2020-01-12:
As I checked lately, ini strings for a game could be quite long if all ROM bits are listed. 
And you still have to create an INI or convert MRA to INI. So instead of an MRA2INI, an MRA2ROM utility which does the conversion offline would be an acceptable solution I think. ROM files are easy to use, even one can put them into -corename- directory, and "F" option will list them by default.
DIP switches are another question.

jotego, 2020-01-12:
Yes, I agree. the offline tool is probably the best solution.

jotego, 2020-01-16:
I'll ask for help in the Atari forum. I'll try to get some software developer to work on this.

closed 2020-01-16 by jotego

gyurco, 2020-01-16:
I doubt there will be any help in that forum, but good luck ;)