mist-devel/mist-firmware #45: Core path in arc files

eubrunosilva, 2020-03-01:
Hi

i already know that we cannot insert full path for rbf in arc files.

but its possible to firmware to check in current folder and also in root folder. Example:

for mcr3mono I have:
Arcade/*.arc and *.rom
Arcade/Alternatives/Rampage/*.arc and *.rom
Arcade/Alternatives/Sarge/*.arc and *.rom

if I must have rbf in arc folder i must have a copy of rbf in each folder
but if I could have the rbf in root folder, I only need one rbf in root folder (this way is easy to maintain)

Thanks

gyurco, 2020-03-04:
Fallback to the root is possible, but don't know if it wouldn't be confusing.

eubrunosilva, 2020-03-04:
ok. think about it.

me for and other would improve organization (i know its against what i was asking not to have cores in root but i prefer to have cores in root and only have one rbf than several :) )

thanks

gyurco, 2020-03-04:
And why it's good to separate the mods from the main ROMs? If you create a Ramapage folder, then you can put all the ROMs and ARCs there, including the official and the modified ones, too.

eubrunosilva, 2020-03-04:
yes but the goal (at least mine) is to have all "oficial/parent" games in one folder, and the hacks/versions of the games in others

gyurco, 2020-03-05:
That's ok, everyone have his/her own view. Including me, which says I'm not sure it should be done this way :)

jotego, 2021-05-24:
Could you give this a second thought, Gyurco? A common folder for ROM files is certainly useful for development. It lets you have different RBF files in different folders, each folder can have a copy of the same ARC file. But the heavy .ROM files are all kept in a common place, say /ROM.

The default can be to check in the current folder, and if that fails, look in /ROM.

The use case is this:

-You have a new bug, introduced by some change in the code
-You can build the core at different commits, and store each one in its own folder
-You can start each core from its own subfolder, the large ROM files are kept in a single location. You can find the commit that broke the code this way

Of course, some users would just like this because it's easier to have all ROMs together and then they can have different folders that organize ARC files in different ways

-Alphabetical
-Genre
-Year
-Etc.

gyurco, 2021-05-24:
_The default can be to check in the current folder, and if that fails, look in /ROM._

As I wrote on the forum, this is already in place.

jotego, 2021-05-24:
I have verified that behaviour just now. It falls back to the root folder. That's good enough for development.

jotego, 2023-11-02:
Maybe this one can be closed...

jotego, 2024-04-13:
Checking on this again, the ROM file gets loaded from the root folder but the RBF file is only checked in the folder where the ARC was called from. Could we have the firmware check the root folder for the RBF too?

I'd like to be able to replicate the MiSTer core display of main sets and then alternative versions in an `alternatives/game-name/many ARC`. Having the ROM and the RBF at root is no problem because you can always hide them using the file attribute. But having to duplicate the RBF is not great.

Could you consider supporting RBF load from root when the ARC is not at root?

gyurco, 2024-04-13:
It's possible to add root as a fallback option. Or a preconfigured directory...but then another config option to remember. Maybe root is enough.

jotego, 2024-04-13:
Yes, root should be enough.

gyurco, 2024-04-13:
Here's a firmware for SiDi128 to try.
[firmware-sidi128.zip](https://github.com/mist-devel/mist-firmware/files/14968226/firmware-sidi128.zip)

uchristo, 2026-10-21:
Out of curiosity… has this approach (roms in one folder, rbfs in another) found it‘s way to the master branch?

Defining seperate rbf and rompaths im mist.ini would be great for not having the same rbf laying around multiple times in different subfolders. One. Would just need to sort the arc-files and each arc would fall back to the root/rbf-cores and root/roms or however you‘d define the folder in mist.ini