mist-devel/mist-firmware #46: Any limits about CONF in ARC files?

jotego, 2020-04-13:
The OSD seems to get broken with this ARC file. I wonder if there is any limit in max number of characters or allowed characters in the string:

```[ARC]
RBF=JTCPS1
MOD=114
NAME=AREA88
CONF="O02,Coin A,2 Coins/1 Credit (1 to continue),4 Coins/1 Credit,3 Coins/1 Credit,2 Coins/1 Credit,1 Coin/4 Credits,1 Coin/3 Credits,1 Coin/2 Credits,1 Coin/1 Credit"
CONF="O35,Coin B,2 Coins/1 Credit (1 to continue),4 Coins/1 Credit,3 Coins/1 Credit,2 Coins/1 Credit,1 Coin/4 Credits,1 Coin/3 Credits,1 Coin/2 Credits,1 Coin/1 Credit"
CONF="O6,Unused,On,Off"
CONF="O7,Unused,On,Off"
CONF="O8A,Difficulty,7 (Hardest),6,5,4,3 (Normal),2,1,0 (Easiest)"
CONF="OBC,Damage,Biggest,Big,Small,Normal"
CONF="OD,Unused,On,Off"
CONF="OE,Unused,On,Off"
CONF="OF,Unused,On,Off"
CONF="OG,Unused,On,Off"
CONF="OH,Unused,On,Off"
CONF="OI,Free Play,On,Off"
CONF="OJ,Freeze,On,Off"
CONF="OK,Flip Screen,On,Off"
CONF="OL,Demo Sounds,On,Off"
CONF="OM,Allow Continue,Yes,No"
CONF="ON,Game Mode,Test,Game"
```

What am I missing?

gyurco, 2020-04-13:
Yes, there's a limit of 256 bytes.
https://github.com/mist-devel/mist-firmware/blob/master/arc_file.c#L9

Personally I think the Unused options just clutter the OSD, they should not be there.
Also this one: 2 Coins/1 Credit (1 to continue) looks too long (OSD horizontal size is 28 chars).

gyurco, 2020-04-13:
+ don't use O0, because it's the ARM reset.

jotego, 2020-04-13:
So the data is sent in the status word then... In MiSTer the solution was to use a different ioctl_index and get it through the ROM interface. That has the benefit of avoing overlapping bits.

I think I understand it now. Thank you!

gyurco, 2020-04-13:
Yes. But it was already discussed here:
https://github.com/mist-devel/mist-firmware/issues/42#issuecomment-591898978

jotego, 2020-04-13:
I think that limit you pointed out refers to the whole config string. I am seeing problem with those long first two lines. I suppose there is a second limit to the maximum length of an entry. You mentioned 28 as the size of the OSD, but maybe I should keep my entries to that length or so.

jotego, 2020-04-13:
I think I see it now: I have to include all the extra components in the config string. (The O12,O34... the commas...)

closed 2020-04-13 by jotego