mist-devel/mist-firmware #47: Default value for config string in ARC?

jotego, 2020-04-13:
Is there a way to provide a default value to the bits defined in the config string? I have a nasty case with CPS1 because there is a freeze bit that freezes the game and it gets enabled by default at the moment.

I cannot invert it in the core because the position is not fixed across the many games.

This forces the user to start up the game and look for the freeze option manually. I can imagine many people asking why the game is frozen.

gyurco, 2020-04-13:
Current workaround can be to supply a CORENAME.CFG file with the desired 32 bit value.
Maybe a DEFAULT=<32 bit hex value> can be added to the ARC file.

jotego, 2020-04-13:
Adding the entry to the ARC file is the best option, but I can use the workaround to get out of trouble.

Thanks!

closed 2020-04-13 by jotego