mist-devel/mist-firmware #48: ARC file config string limit too short

jotego, 2020-04-13:
I have been removing a lot of text from the DIP switch definition, but I still run over 256 bytes. In general I am getting between 300 and 400 characters. CPS1 has three DIP switches rather than the usual two, so that doesn't help either. Could we rise the limit to 512 bytes? I understand the microcontroller is constrained in memory and this may not be possible.

gyurco, 2020-04-13:
It still can compile with 512, so OK from my side. Then I'll add default config value to ARC before releasing a new firmware. It would be good to convenience @sebdel also to add this value from the MRA (the switches tag could have a default attribute).

jotego, 2020-04-13:
Thank you. I'll ask him.

sebdel, 2020-04-13:
Thanks to the new github app, I get notifications when I'm mentioned. Neat. Please file a PR @jotego , I didn't understand what's needed. It's about string size or default values? Or both?

jotego, 2020-04-17:
I have the [PR ready now](https://github.com/sebdel/mra-tools-c/pull/65).
This is a sample output
```
[ARC]
RBF=JTCPS1
MOD=114
NAME=WILLOW
DEFAULT=FFABFF
CONF="OP,Freeze,On,Off"
CONF="OS,Flip Screen,On,Off"
CONF="OT,Demo Sounds,Off,On"
CONF="OV,Game Mode,Test,Game"
```
I agree to limit DIP switches from bits 23 down to 8 in MiST. Let's leave the first 8 bits for system-level options.

gyurco, 2020-04-17:
Added DEFAULT= to firmware
Note that the number has to be prefixed with 0x to be interpreted as HEX value. Otherwise it will be handled as decimal.

commit e648509b (2020-04-17): ARC: add DEFAULT setting

Fixes #48
closed 2020-04-17 by gyurco