mist-devel/mist-firmware #49: Discussion: separate ioctl_index for DIP switches

jotego, 2020-04-21:
I know this was partly discussed [here](https://github.com/mist-devel/mist-firmware/issues/42), but now that I have gone through this process for three games, I keep thinking that separating switches from the status word has value:

1. Larger switch width, MiSTer uses 64 bits
2. Keep MiST system OSD settings separate from game specific settings. Currently I am leaving 12 bits for OSD settings and the rest for core specific, but this is not enough for all games. Plus a bit of a headache to maintain

I know other developers have already used the new ARC/DIP files. So they'll sure have an opinion on this topic I'd like to hear.

gyurco, 2020-04-21:
I think the only other developer used was me :)
Do you really need this large number of switches? With a single OSD screen, the huge number of options are not looking really nice (2 years ago it was limited to 7 lines).

sorgelig, 2020-04-21:
O,T,R - for 0..31
o,t,r - for 32..63

jotego, 2020-04-21:
I agree that many options in such a small OSD won't look great. MiSTer take on that has been to move DIP switch settings to a second screen, so system OSD options appear on the first screen, and from it you can select a "DIP switches" item to move onto the second. Certainly this design option is independent of the whether the status word is used or not.

Why is it hard to drop DIP switches out of the OSD menu?

1. DIP options may not be fully documented in MAME. Example: Legendary Wings, where two bits are marked as Unknown but if you go into the game test mode, their functionality is written in it. 
2. DIP switches may have functionality for test, debug and even cheat purpose; which is reported in some websites and hard core players know about. Example: the starting level of some CAPCOM games depend on the value of the test and coin/credit switches. SF2 has many hidden debug screens.
3. DIP switches may affect core behaviour in undocumented ways. Example: the starting sequence of GunSmoke depends on the value of a supposedly unrelated DIP switch.

So it is not so easy to let DIP switches out. Eventually someone comes back and ask for the missing switch!

gyurco, 2020-12-20:
From firmware_201220, the status word is 64-bit.

closed 2020-12-20 by gyurco