mist-devel/mist-firmware #50: RBF not loaded if MOD is not in ARC file

jotego, 2020-04-21:
I had problems yesterday when trying to use ARC files for cores that didn't need a MOD byte. Omitting MOD caused the core loading to fail.
I think cores should load from ARC even if the MOD line is ommitted. Note that the MRA tool will create ARC files without MOD if no rom index=1 is especified in the mra file.

gyurco, 2020-04-21:
That's somewhat intended, as missing MOD value indicates an error to the ARC loader. You can put MOD=0 if it's missing. I think it would be easy to modify the MRA tool to put it if not rom index=1 found.

jotego, 2020-04-21:
I have modified the MRA tool for this.
https://github.com/sebdel/mra-tools-c/pull/68

closed 2020-04-21 by jotego