mist-devel/mist-firmware #51: WiFi SD card don't work.

xolod79, 2020-04-28:
I tried [PQI Air WiFi SD Card](https://forum.archive.openwrt.org/viewtopic.php?id=45820&p=10#p297806) , but don't work. 

Terminal output MIST

Version ATH200215
                                                
SDHC card detected
spiclk: 24 MHz
usb_init
max3421e_init()
Chip revision: 13
busprobe()
usb_reset_state
Fatal error: 2

I think this card does not have enough time to initialize, since there is micro Linux inside it.
In source mmc.c I found this:

   _timeout = GetTimer(1000); // initialization timeout 1s, 4s doesn't work with the original arm timer_

Let's try to increase the delay. If it does not bother you, please build the firmware to debug this problem.

gyurco, 2020-04-28:
Did you try to compile it yourself?
https://github.com/mist-devel/mist-board/wiki/HowToCompileTheFirmware

xolod79, 2020-04-28:
I will try to build.
What maximum delay can I set?

gyurco, 2020-04-28:
I have no idea, never touched that value.

sebdel, 2020-04-28:
@xolod79, don't build your own gcc. It takes forever. Just install docker if you don't already have it, git clone the firmware and do this at the root dir:
```
docker run --rm -v $(pwd):/build st3ph4n/gcc-arm-none-eabi make
```
That's it, your own firmware in a matter of seconds.

xolod79, 2020-04-28:
@sebdel thanks for the good advice, but gcc is already built!
I decided to insert a lot of dumb printf into the mmc.c initialization procedure SD Card.

// init memory card
unsigned char MMC_Init(void)
{
    unsigned char n;
    unsigned char ocr[4];

    if(!mmc_inserted()) {
      iprintf("No card inserted\r");
      return(CARDTYPE_NONE);
    }

    spi_slow();     // set slow clock
    DisableCard();  // CS = 1
    SPI(0xff);      // DI = 1
    TIMER_wait(20);  // 20ms delay
    for (n=0; n<10; n++) SPI(0xff); // 80 dummy clocks, DI = 1
    TIMER_wait(20);  // 20ms delay
    EnableCard();

    CardType = CARDTYPE_NONE;

      iprintf("***### My Laber 1 ###****\r");

    for(n=0; n<16; n++) {
      TIMER_wait(1);

      iprintf("***### My Laber 2 ###****\r");

      if (MMC_Command(CMD0, 0) == 0x01) break; // try to send CMD0 multiple times
    }
    if (n<16) // got CMD0 IDLE response
    { // idle state

      iprintf("***### My Laber 3 ###****\r");

        timeout = GetTimer(1000); // initialization timeout 1s, 4s doesn't work with the original arm timer
        if (MMC_Command(CMD8, 0x1AA) == 0x01) // check if the card can operate with 2.7-3.6V power
        {   // SDHC card
            for (n = 0; n < 4; n++)
                ocr[n] = SPI(0xFF); // get the rest of R7 response
            if (ocr[2] == 0x01 && ocr[3] == 0xAA)
            { // the card can work at 2.7-3.6V

      iprintf("***### My Laber 4 ###****\r");

                iprintf("SDHC card detected\r");
                while (!CheckTimer(timeout))
                { // now we must wait until CMD41 returns 0 (or timeout elapses)

      iprintf("***### My Laber 5 ###****\r");

                    if (MMC_Command(CMD55, 0) == 0x01)
                    { // CMD55 must precede any ACMD command

      iprintf("***### My Laber 6 ###****\r");

                        if (MMC_Command(CMD41, 1 << 30) == 0x00) // ACMD41 with HCS bit
                        { // initialization completed

      iprintf("***### My Laber 7 ###****\r");

                            if (MMC_Command(CMD58, 0) == 0x00)
                            { // check CCS (Card Capacity Status) bit in the OCR
                                for (n = 0; n < 4; n++)
                                    ocr[n] = SPI(0xFF);
                                CardType = (ocr[0] & 0x40) ? CARDTYPE_SDHC : CARDTYPE_SD; // if CCS set then the card is SDHC compatible

[... 155 lines cut ...]
***### My Laber 1 ###***
***### My Laber 2 ###***
***### My Laber 2 ###***
***### My Laber 2 ###***
***### My Laber 2 ###***
***### My Laber 2 ###***
***### My Laber 2 ###***
***### My Laber 2 ###***
***### My Laber 2 ###***
***### My Laber 2 ###***
***### My Laber 2 ###***
***### My Laber 2 ###***
***### My Laber 2 ###***
***### My Laber 2 ###***
***### My Laber 2 ###***
***### My Laber 2 ###***
***### My Laber 2 ###***
No memory card detected!                                                        
spiclk: 24 MHz                                                                  
usb_init                                                                        
max3421e_init()                                                                 
Chip revision: 13                                                               
busprobe()                                                                      
usb_reset_state                                                                 
Fatal error: 1

xolod79, 2020-04-29:
Second attempt:
I found that the problem happens in the MMC_Read procedure in mmc.c
he giving out a message
CMD17 (READ_BLOCK): invalid response 0x80 (lba = 0)
added a delay of 300ms forward neither, but didn’t help.

gyurco, 2020-04-29:
0x80 is invalid indeed. I'm curious what's the problem, if you can found out.

xolod79, 2020-04-30:
My friend found this message exactly with this behavior (respons 0x80) when working in SPI mode.
https://forum.archive.openwrt.org/viewtopic.php?id=45820&p=6   (post 136)
Most likely this card does not know how to work in SPI mode.
Which confirms another Japanese source.
https://blog.babibubebo.org/archives/676
They checked:
Eye-Fi  - Support SPI
FlashAir - Support SPI
Flucard PRO - Don't support SPI
PQI Air Card - Don't support SPI
ezShare - Don't support SPI
Transcend WiFi SD Card  - Don't support SPI
QlixSynchro - Unknown
CloudFlash - Unknown
Sky Share S10 - Unknown

gyurco, 2020-04-30:
That means it cannot be solved then.

sorgelig, 2020-05-01:
I see PQI firmware is heavily modded, so may be someone will improve the firmware to support SPI.
But as far as i know, these WiFi cards support only reading over WiFi. You can't upload anything back to SD card as it will eventually tamper file system. Local user (i.e MiST in this case) won't see the changes (or will see it incorrectly) as it doesn't expect changes on SD card on-the-fly. And if both MiST and WiFi will write to card, it will become corrupted.

xolod79, 2020-05-01:
As far as I know, the source code of the ka2000_sdio.ko and ka2000_sdhc.ko drivers is not available, which makes their development a difficult task.
I also suggest that there may be hardware limitations for spi mode for this card.
Since this kind of card is working under Linux, it goes to it for reading and writing.
Yes, to be sure, the consistency of the file system will be violated if it is recorded on it from the Linux OS and from the host system (for example MIST) where the card is inserted.
But this can be avoided if you first write the necessary files via wifi to the card, restart MIST, he re-reads the FS structure and then you can use the recorded files.

sorgelig, 2020-05-01:
Probably, only modded firmwares allow to write over WiFi. So most likely all the rest WiFi SD cards from list above support only reading over WiFi. I have Toshiba FlashAir - it supports only reading.

xolod79, 2020-05-01:
Yes, I am litle hack my card. Started telnet, ftp and copy files over ftp.

sorgelig, 2020-05-02:
In theory, such card if WiFi write detected, could immediately set SD side as read-only till next initialization. This would prevent corruption from accidental write from MiST side. 
But since it doesn't support SPI mode, it's in-vain. Such card could be used in MiSTer as secondary SD card. But it also requires the SPI mode. Can work as primary SD card, but WiFi there is useless.

xolod79, 2020-05-06:
I checked what will happen after I recorded via WiFi on an SD card. Access to SD from the host to write is not blocked.
And recording files from the host will overwrite the files recorded by the WIFI caser. Since the host has nothing to know about changes in the file system =)

gyurco, 2020-05-07:
If you're lucky.
If you're unlucky, it can destroy significant part of the file system.
If it would work on MiST, then you must be sure that no writes happen simultaneously, and reboot the ARM controller after you finish your copy job.

xolod79, 2020-05-07:
yes totally agree

harbaum, 2020-05-07:
This could be solved by letting the PC write some magic file or the like and having the MIST frequently check for the presence of that file. If it detects the file it reboots itself and deletes the file. That way this could be handled fully automatic. But this still has the risk of simultaneous writes and data corruption.

sorgelig, 2020-05-07:
Better to mod SD firmware to immediately make it read-only (whole SD card) once WiFi write is detected. Then reboot (with re-init SPI mode) will disable read-only mode.
But the main problem - is inability to work in SPI mode.

harbaum, 2020-05-07:
And how's the board supposed to understand that a reboot is due?

sorgelig, 2020-05-07:
probably by issuing initialization commands. I don't remember right now which exactly, but there are mandatory initial commands for SD.

gyurco, 2020-05-07:
The question was how the MiST board would know if it should reboot itself.
Btw, the monitoring file should have two states:
- external (WiFI access) in progress
- external access finished
This is to not reboot during writing, but just after the writing is done.

sorgelig, 2020-05-07:
> The question was how the MiST board would know if it should reboot itself.

MiST doesn't have to know/react immediately. RO mode is for safety. After uploading the files, user manually can press reset button on MiST. It's much easier than manually upload the file after each change. Especially while big file uploading MiST may modify FS.

closed 2020-06-21 by xolod79