mist-devel/mist-firmware #54: Tile-mapped OSD?

jotego, 2020-05-14:
The current OSD is a bitmap. We could apply the traditional tile map technology to obtain some benefits:

* Smaller memory foot print for the window size
* Variable fonts
* Larger screen size
* Hardware scrolling for each row
* 4 indexed colours

To fill a 32 row x 32 column image (256x256 pixels) using a 9-bit code per 8x8 tile takes:

-1kB for the tile map
-1kB for a 128-character 1 bit-per-pixel font
-Bits 6-0 would be used for code, bits 8-7 for colour. Note that BRAM blocks are 9-bit wide (not 8) so we take advantage of the extra bit

By using the first byte of each row as a scroll value, it would be possible to provide individual row scrolling.

The firmware can be compatible with this style of OSD and the old one. So no need for updating old cores although ideally a module swap should be enough.

If you find the idea worth pursuing I'm willing to write the verilog side of this. What do you think?

gyurco, 2020-05-14:
The actual OSD rendering is basically based on tilemap - strings rendered with a font.
What you suggest is to share the rendering job between the MCU and the FPGA, which might be an interesting option. Just the questions about the details:
- Where the tile data (font + other gadgets) should come from? I guess it should be defined on the FPGA side.
- HW scrolling: should be row based? But then the MCU should still do the scroll, and send an xscroll value for each row. However, scrolling currently is not row based,e.g. the left side of the OSD is not scrolled. Defining the scrolling region must be done wisely.
- Color code: from the tilemap (sent by the MCU) or in tile data?
- Size change: for this, the menus and other places should be updated, too, now supporting two different OSD dimensions.

jotego, 2020-05-14:
Let me reply first:

* Tile data is sent via SPI to the OSD module. MiSTer has loads of fonts that could be used, even letting the user choose a font by placing a file or another in the SD card
* About row scrolling: making it windowed does complicate things
* If tile data is monochrome, the could would have to go in the map data, where two bits could choose betwee four colors to apply to the whole tile
* Yes, the OSD elements will need redesigning for a larger size

This is quite a change so we must be sure of what the gains would be:

Pros
* Nice color fonts, with plenty of space for options
* Nicer user experience
* An interface look&feel change sends the message "we are alive" to the world. That states continuity of the project
* For the same screen size, less memory resources in FPGA

Cons
* Writting new OSD (both verilog and firmware sides) will take some time off other topics
* Does not apply to old cores unless someone goes and updates them (I'm ok with updating JT titles)
* Firmware to support two OSD variants

harbaum, 2020-05-14:
The current OSD is basically just a 2kb bitmap buffer. A single 8*8 font with 128 chars is 1k. And you still need a small tile buffer for 8*32 characters. You can save a few bytes that way. But you'll need all the intelligence for normal and inverted rendering, the scrolling and the rotated text on the left side on the fpga.

Plus you might not save any memory at all as the bitmap buffer fits into a single block ram block. Using separate font and tilemap memory may use two blocks. So you might actually increase memory consumption.

gyurco, 2020-05-14:
There's already a system.fnt loading for the old MiST core, that can be extended to load the OSD font, too (even without changing the current rendering method).
Yes, implementing this need some enthusiasm, which I'm currently lacking about pimping up the OSD :)

gyurco, 2020-05-14:
However I'm wondering why this is used as the OSD buffer, it's 16kbits:
reg  [7:0] osd_buffer[2047:0];

sorgelig, 2020-05-14:
Pros 1-3 are basically the same thing. Not always multicolor OSD is nice. One thing is when OSD completely redesigned with per-pixel transparency and at least 256 colors. So you will really have new OSD. But proposed change gives no real advantage. 3 additional colors per tile? It will be noticed just for a couple days. It won't give anything new. Only some devs will add ugly looking cacophony of colors. It won't add information.
Scrolling lines currently has unlimited length with pretty much easy management from MCU. And it's easy to define which part of OSD should be scrolled. New way will be kind of mini-NES with sprites/tiles which should be smartly controlled by MCU. You will need to construct a larger tile map with controlled visible part, so you will be able to scroll it.

A lot of change for doubtful advantage.

As for font customization: it can be done for MiST through special tool which will replace the part in FW (marked specifically in advance). So using this tool people can inject the font they like and then update the firmware.

gyurco, 2020-05-14:
_An interface look&feel change sends the message "we are alive" to the world. That states continuity of the project_
Yeah, nowadays companies are using this to sell the same shit again and again. Maybe it could be done for the MiST, too, but I was never good in marketing :)

jotego, 2020-05-14:
Thank you for your inputs. Yes, I see that there is little benefit for the ammount of work.

closed 2020-05-14 by jotego

gyurco, 2020-05-14:
I think I will add the font loading, it's just moving it from tos.c to a main place.

jotego, 2020-05-14:
Good!

harbaum, 2020-05-14:
Plus you can do all the eye candy in the fpga anyway. You can do multicolored shadow effects if you want. You can give different parts different colors. You can even add a core specific logo.

jotego, 2020-05-14:
> Plus you can do all the eye candy in the fpga anyway. You can do multicolored shadow effects if you want. You can give different parts different colors. You can even add a core specific logo.

You mean, with the current solution, right?

harbaum, 2020-05-14:
> You mean, with the current solution, right?

Yes, sure. You just need to be able to process the bitmap somehow. But you can display it however you want with all the eye-candy you come up with.

gyurco, 2020-05-14:
@jotego already did with the background of his OSD :)

Btw, the SYSTEM.FNT support in tos.c is for 8x16 fonts. Is there any reason to keep it? If such file is not found, the internal 8x8 font is upscaled. I never had a 8x16 SYSTEM.FNT file, and that's only used for messages you probably won't ever see (at boot of the ST core).
My plan is to move only a common 8x8 font support for the old ST core boot, Minimig boot and OSD.

harbaum, 2020-05-14:
Feel free to remove that. That was actually one of the first things I ever coded in the firmware. There was no running core and I wanted to see something on the screen. It only works in st high resolution, anyway. This really isn't needed anymore.

gyurco, 2020-05-14:
8x8 font loading support is added. Interesting to see the OSD with the ZX Spectrum font.