mist-devel/mist-firmware #56: [mist_video] menu size changes with `.ce_divider()`

nippur72, 2020-06-20:
in `mist_video.v`, the horizontal width of the OSD menu is different when having:

```
.ce_divider(1),   // and 2x pixel clock => normally large menu

// vs

.ce_divider(0),   // and 4x pixel clock => smaller menu
```
If I am not wrong, the size should be the same, right?

harbaum, 2020-06-20:
What did you expect? If you increase the clock then the pixels are shifted out faster and the OSD becomes narrower.

nippur72, 2020-06-20:
the `OSD` has already a logic to determine the pixelsize (see [here](https://github.com/mist-devel/mist-modules/blob/master/osd.v#L118)), perhaps it could double the size when using the 4x clock.

P.S. sorry this issue was meant to for `mist-modules` not `mist-firmware` 😅

gyurco, 2020-06-20:
Probably with 4x pixel clock, you already reach pixsz=3. But I think you should be using very high video clocks then.

closed 2020-06-22 by nippur72