mist-devel/mist-firmware #68: long initial boot time with Amiga hdf = 14GB

gcopoix, 2021-06-08:
With new exFAT Support I've tried a 14GB HDF File on a 32GB SD Card.
Works basically great (thanks for supporting exFAT), but on initial boot (start up after loading core) it takes appr. 65 seconds (time between MiniMIG splash and SD access shown by red LED) until booting up.
A soft-reset or reset by F12  will boot up as expected (takes 1-3s until it starts booting, expected time known from Amiga).
Any reason for this "initial delay"?

gyurco, 2021-06-08:
Maybe indexing if the hdf is too fragmented.

gcopoix, 2021-06-08:
red LED is off during this time

gyurco, 2021-06-08:
It's normal, indexing doesn't turn on the LED (nor config file reads - don't tell anyone).
You can use bigger cluster size (exFAT supports very big cluster sizes), also defragment the file, then it should take much less FAT reads.

harbaum, 2021-06-08:
A 14GB file? What cluster size does that use? Whatever ... Very impressive. 

I've only tested the new FS driver a few times but had no issues so far.

Bzw: The real test is not running this firmware or updating to it but updating from it. This is the critical test as it proves that all routines required for an update are properly moved to ram and are available even after the flash has been erased in preparation of the update to be written.

You can e.g. try to downgrade from this new version to a previous one. And try to catch corner cases during that like crossing cluster boundaries or the like.

gyurco, 2021-06-08:
With a 64K cluster size, it's 229000 clusters :)
You can even get a 32GB hardfile:
http://coffin.ninex.info/
I have no idea what's inside, I didn't want to reformat my existing FAT32 SD Card to try.

I tried firmware updating a few times, also there's a modified f_read in fat_compat.c (FileReadNextBlock) which tries hard to not use any library functions. I hope it's OK.
One thing I don't understand: why ConfigureFpga has to be a RAMFUNC?
https://github.com/mist-devel/mist-firmware/blob/master/fpga.c#L248

harbaum, 2021-06-08:
Have you tried removing the RAMFUNC decorator from that? It seems to call other functions which are not in RAM. So this won't work with erased flash anyway 

I honestly don't remember why I did that ... nine years ago 😀

Anyway, may have to do with timing? Maybe code from ram runs faster? I don't know ...

harbaum, 2021-06-08:
Uhm ... that's also in the Xilinx code. So I likely took that from the Minimig without questioning it.

This was one of the first routines i touched. So I may just have kept such things which I didn't understand as long as I tried to get something to work at all ...

gyurco, 2021-06-08:
Yes, I tried, nothing broke.
Maybe another forgotten story (I found a lot of interesting legacy things during the FS update).

gcopoix, 2021-06-09:
Tried with different cluster sizes. With 32k or 64k cluster size it is "only" appr. 10s for indexing, which is ok for me.
My initial try was with 2k clustersize, probably not the best idea.
Maybe worth a comment on SD card setup.
Thanks

closed 2021-06-09 by gcopoix