mist-devel/mist-firmware #73: core.arc doesn't work anymore

jotego, 2021-07-16:
In one of the recent firmware updates, `core.arc` as the boot core has stopped working. This fails at least on _firmware_210618_ .

closed 2021-07-23 by gyurco

jotego, 2021-07-25:
Thank you for solving this. Could you upload a binary, please?

gyurco, 2021-07-25:
Jose, here's one. There's still a strangeness, when you're uploading via USB-Blaster, the CORE.ARC file is loaded, and the wrong ROM is opened. But it won't affect normal use.
[fw-210725.zip](https://github.com/mist-devel/mist-firmware/files/6873845/fw-210725.zip)

jotego, 2021-07-25:
Thank you so much. I installed it right away. I think the problem of not loading the ROM when uploading via USB-Blaster was present before. If you ever have time to fix it, it'd make my developer life a little better :-)

gyurco, 2021-07-27:
(Hopefully) fixed in today's release.

commit e20cb1ab (2021-07-23): Fix core.arc loading

Fixes: #73