mist-devel/mist-firmware #79: disable SDHC warning in user_io

rampa069, 2021-09-03:
When using virtual images, user_io is detecting the **real** card model in the host and displaying a warning on each disk access. I think is safe to delete this errorlog.

```--- a/user_io.c
+++ b/user_io.c
@@ -1358,10 +1358,10 @@ void user_io_poll() {
                                        if(using_sdhc) {
                                                // we have not been using sdhc so far? 
                                                // -> complain!
-                                               ErrorMessage(" This core does not support\n"
-                                                            " SDHC cards. Using them may\n"
-                                                            " lead to data corruption.\n\n"
-                                                           " Please use an SD card <2GB!", 0);
+                                               //ErrorMessage(" This core does not support\n"
+                                               //             " SDHC cards. Using them may\n"
+                                               //             " lead to data corruption.\n\n"
+                                               //            " Please use an SD card <2GB!", 0);
                                                using_sdhc = 0;
```

gyurco, 2021-09-03:
The problem is that the sd_card.v supplies the OCR high capacity bit from the original SD Card, even in  a .VHD case. And if the core doesn't aware of it, it might supply byte count instead of LBA for the READ/WRITE commands, which won't be translated to LBA -> data corruption will happen.
Actually you only need to supply '1' to sd_sdhc in user_io to suppress this warning (and make sure the core/firmware in the core can handle the high capacity cards correctly).

rampa069, 2021-09-03:
no... i am using vhd beacause the core does not how to handle SDHC (only low level SDC commands) so if i fore the sdhc mode, i got no warnings, but no disk operations is working.... I All fixed if all the vhd are copied to a  SDC but they are rare now a days..... (and expensive)

rampa069, 2021-09-03:
i have two cores at the moment with this problem... multi-cpm and memotech MTX512. I can distribute a modified .upg firmware with them, but i think this isnt the correct way....

gyurco, 2021-09-03:
_only low level SDC commands_

That means the sd_card.v wrapper is not used, but only the sd_xx signals in user_io?
Or can you show me the source code? Suppressing the warning is just hiding a bug somewhere.

gyurco, 2021-09-06:
Can you try the attached sd_card.v?
You can give it allow_sdhc = 0, and it still should report sdhc use to the firmware in case of a VHD. The warning will displayed only in a pass-through case.
[sd_card.v.zip](https://github.com/mist-devel/mist-firmware/files/7117497/sd_card.v.zip)

rampa069, 2021-09-07:
it worked. thanks!

gyurco, 2021-09-08:
Great! I added it to mist-modules (with a small bugfix).

closed 2021-09-08 by gyurco