mist-devel/mist-firmware #81: Retrobit Mega Drive Controller USB issues

sprudel79, 2021-12-11:
Hi there,

with firmware firmware_120721 my Retrobit Mega Drive Controller USB doesn't work anymore unfortunately. Two firmware releases ago it was detected correctly thanks to your work.
Now it is recognized as Retrobit NES Controller and the button setup is messed up.
Could you please verify? I can support testing if required.
Thank you and regards,
Sprudel

gyurco, 2021-12-11:
What's the USB VID/PID? And what was before? If it's recognized as a different controller now, then it must be changed.

sprudel79, 2021-12-12:
Thanks for your quick reply, I will check the details today evening and let you know.

sprudel79, 2021-12-12:
Hi @gyurco, I was able to check things with my second MIST FPGA which had the firmware 211001. The Retrobit Genesis 6-button Controller USB was detected as: 0F0D 00C1.
Next I have upgraded to firmware 211207 and rebooted. The controller was then detected aa: 0079 0011 so yes indeed the VID/PID have changed.

gyurco, 2021-12-12:
This gamepad has the weirdest firmware ever.
Can you try the attached test firmware?
[fw-test-211212.zip](https://github.com/mist-devel/mist-firmware/files/7699322/fw-test-211212.zip)

sprudel79, 2021-12-12:
Thank you very much. I was able to test it.
The VID/PID are still the same and it still says Retrobit NES but more buttons are working.
With a custom joy map setup in the mist.ini I am able to use the controller without any problem anymore. Thank you very much again.

gyurco, 2021-12-12:
Maybe the gamepad can be switched to Sega/Nintendo mode via a button press during power up.

sprudel79, 2021-12-12:
To be honest I don't care what the label says :-) As long as the buttons are detected I am fine. And thanks to the great flexibility via mapping I can easily achieve to have a perfect setup.
Thank you and everyone else involved in the firmware and core development for your efforts and great work!