mist-devel/mist-firmware #82: Xbox 360 Style Game Controller does no longer work with firmware 211207

Negsol, 2021-12-20:
Easy SMX SL-9111 Gaming Controller an Xbox 360 Style Game Controller does no longer work with firmware 211207. Worked fine with firmware 211001 and before.

Under firmware 211001:
USB vendor ID: 11C0, product ID: 9107
[firmware_211001.txt](https://github.com/mist-devel/mist-firmware/files/7746277/firmware_211001.txt)

Under firmware 211207:
HID: unsupported descriptor type 65 size 6
[firmware_211207.txt](https://github.com/mist-devel/mist-firmware/files/7746278/firmware_211207.txt)

gyurco, 2021-12-20:
But later on:
HID: usb_hid_init(2)
HID: unsupported descriptor type 65 size 6
HID: no hid interfaces found
 -> not accepted :-(
trying to init class 2
 -> accepted :-)

So it was recognized as an XBOX controller.
On the previous firmware, it was a HID controller with different PID/VID. Usually this change is done by pressing a button when powering up.
So it doesn't work as XBOX controller? What's on the input test screen?

Negsol, 2021-12-20:
With the 211207 firmware there is nothing under Joy1

Joy1 -
Joy2 - Atari DB0 Joystick
...

With the old firmware it was USBID 11C0 and PID 9107

---
I know that a lot of controller can switch between different modes. This one however does not - I upload the manual pages I have
[Manual_SL9111.zip](https://github.com/mist-devel/mist-firmware/files/7746457/Manual_SL9111.zip)

So basically nothing happens when I press the home key for 5 secs (mentioned in manual for reconnecting). Also worthy of note it is wired controller.

Negsol, 2021-12-20:
Also it worked before for over 12 months now (it is my primary controller for MiST).

gyurco, 2021-12-20:
And if you plug in to a computer, then what mode comes up? I wonder how the MiST firmware could force it to switch modes...
Maybe you can try this test:
https://github.com/mist-devel/mist-firmware/issues/81#issuecomment-991947552

Negsol, 2021-12-20:
Thank you for your suggestion and support in this!

Under Windows it is found as "Xbox 360 Controller For Windows". Everything and all buttons do work here.

Hardware IDs: USB\VID_045E&PID_028E&REV_0115
Compatible Ids: USB\Class_FF&SubClass_5D&Prot_01
Class: XnaComposite

The Test firmware 211212 did nothing for this controller. Everything is the same as with the latest released FW.

gyurco, 2021-12-20:
Then it's also XBOX mode in Windows. Strange, if you reflash the old firmware, it'll go back to HID mode?

Negsol, 2021-12-20:
yes it will - i tested several times switching between the two firmwares (see first post incl. the logs)

gyurco, 2021-12-20:
I have an idea what happens...can you try the attached fw and post the debug output when you plug in this gamepad?
[fw-test-211220.zip](https://github.com/mist-devel/mist-firmware/files/7747270/fw-test-211220.zip)

Negsol, 2021-12-20:
The usb controller works again under MiST with FW 201220! :)

Here is the log (I starter a core to reach the USB Status display. All good
[firmware211220.txt](https://github.com/mist-devel/mist-firmware/files/7747313/firmware211220.txt)
!

Negsol, 2021-12-20:
no I take it back - also with the test firmware it will not work
Let me test again

Negsol, 2021-12-20:
sometimes it requires a few resets or a cold start after a FW update to get a stable situation regarding usb recognition - really strange.

Negsol, 2021-12-20:
so now everything back to square 1
USB controller is not recognized with the test firmeware
here is the log
[firmware211220.txt](https://github.com/mist-devel/mist-firmware/files/7747343/firmware211220.txt)

gyurco, 2021-12-20:
Yes, but I know now why the modes are toggling. Every query of the device descriptor changes the mode. I'll make a change which query the device descriptor only once.

Negsol, 2021-12-20:
thank you - really appreciate your help - as always!! :) I like the controller very much ... I even got a spare one ;) Also the thing could affect other cheap 360 controllers as well ...

gyurco, 2021-12-20:
Here's another test, can you try it please?
[fw-test-211220b.zip](https://github.com/mist-devel/mist-firmware/files/7747399/fw-test-211220b.zip)

Negsol, 2021-12-20:
no change to report - controller still not working 
here is the log
[firmware211220b.txt](https://github.com/mist-devel/mist-firmware/files/7747450/firmware211220b.txt)

gyurco, 2021-12-20:
I expected this - normally the controller initializes in XBOX mode. Then there was an extra query of the device descriptor previously when it probed the device for being a HUB.
Previously:
 - first query (in hub_init): xbox mode
 - second query (in hid_init): HID mode
Then:
- first query (usb_init): xbox mode
- second query (hub_init): HID mode
- third query (hid_init): back to xbox mode
Now:
- first (and only) query: xbox mode

Thus it needs two queries for HID mode. Is manual switching really doesn't work holding some buttons?
I wonder where to add the second query - as its very hacky, and worked previously only by luck.
Or how to fix XBOX mode - that needs debugging the driver.

Negsol, 2021-12-20:
I am not sure I get all of it - but isn't it normal to always send two signals/query to switch back to the original mode? Or is it that all devices have a different amount of modes - then 2 is just luck I guess. Maybe at an option in the mist.ini so that user can at least adjust. hm also am wondering why xbox mode does not work here - again it was a cheap controller so properly not up to specs at all. I have an OG Xbox 360 controller  here, if you need it for me to test something.

gyurco, 2021-12-20:
I'm not sure what's the "normal" way to switch modes, I guess it's device dependent (I never encountered this kind of mode switch, button press during power up is more "standard"). BTW, before I add two queries (which will happen not by luck), here's a test to get some debug info from the XBOX driver. Would be good if you can post the logs of this.
[fw-test-211220c.zip](https://github.com/mist-devel/mist-firmware/files/7747597/fw-test-211220c.zip)

Negsol, 2021-12-20:
No change:
[firmware211220c.txt](https://github.com/mist-devel/mist-firmware/files/7747625/firmware211220c.txt)

Also I am re-reading the manual - Maybe this controller is strange but it is pretty recent and supports the "nintendo switch" - Nowhere in the manual a switch of mode is mentioned other than the use of the home button (holding for 5 secs). However this does absolute nothing for this controller. I have the feeling the controller has to see that it is not connected correctly (resulting in blinking of the LEDs of the controller) then maybe the home button thing can be used. Right now the controller just sit there not switching when I press any of the buttons (or holding down for 5 secs).

gyurco, 2021-12-20:
Ok, so it doesn't even get a valid poll. More debugging info added:
[fw-test-211220d.zip](https://github.com/mist-devel/mist-firmware/files/7747657/fw-test-211220d.zip)

Negsol, 2021-12-20:
I will test it - However the controller is a state (LED 1 and LED2 lit) This state means "connected" - nothing is blinking - The controller is quite happy not doing anything. It nearly looks as if it does not see that the connection is not working. So maybe to smart. Also manual mode change (pressing the mode button) does nothing. LED status remains the same.

On my windows box the driver Xbox 360 is loaded and nothing happens after that.

Negsol, 2021-12-20:
here is the log fil
[firmware211220d.txt](https://github.com/mist-devel/mist-firmware/files/7747705/firmware211220d.txt)
e

Negsol, 2021-12-20:
nothing changed

Negsol, 2021-12-20:
actually on my windows box the mode button toggles different modes

Negsol, 2021-12-20:
It can be switched to somthing recognized as ESM GAMEPAD 1.15

Negsol, 2021-12-20:
HID\VID_045E&PID_028E&IG_00

Negsol, 2021-12-20:
longer pressing the mode key then resets back to the Xbox360 mode. Wonder why this does not work, when it is connected to the MiST.

Negsol, 2021-12-20:
When connected to the mist the mode button loses all functionality

Negsol, 2021-12-20:
Removing the controller from the Mist when powered on gives me the strange debug message "USB: usb_xbox_poll() error: 13"

gyurco, 2021-12-20:
Well, I have no idea why the XBOX mode doesn't work - here's a test with double device descriptor query - it should switch to HID mode.
[fw-test-211220e.zip](https://github.com/mist-devel/mist-firmware/files/7747754/fw-test-211220e.zip)

Negsol, 2021-12-20:
I found it

Negsol, 2021-12-20:
Apparently switching it on a windows box will make the controller remember the last mode. So when I leave it on HID it will remember it also on the MiST.

Negsol, 2021-12-20:
But doing a reset (holding the mode button for 5 secs) brings it back to the xbox 360 mode and it locks up again

Negsol, 2021-12-20:
then turning of and on the MiST will not bring back HID but the controller remains in Xbox 360 mode and the mode switch is blocked.

Negsol, 2021-12-20:
Reconnecting it to the windows box should solve it. But once it gets into xbox mode on the MiST it will remain locked.

Negsol, 2021-12-20:
I will test the other firmware now

Negsol, 2021-12-20:
here is the log:
[firmware211220e.txt](https://github.com/mist-devel/mist-firmware/files/7747802/firmware211220e.txt)

Negsol, 2021-12-20:
So I guess something is forcing the controller into this xbox 360 mode when connected to MiST. This also disables all mode switching on the controller. Under Windows switching works and maybe will be remembered by the controller for some time.

Negsol, 2021-12-20:
This is at least a work around I guess. But always connecting to a different machine just to switch modes is a bit special.

Negsol, 2021-12-20:
I tested it again and it remembers the mode. Outside of the Xbox 380 mode the controller works nicely again under the test firmware!

gyurco, 2021-12-20:
Here's a last test, I wonder if it does the mode switch as in the older ones.
[fw-test-211220f.zip](https://github.com/mist-devel/mist-firmware/files/7747831/fw-test-211220f.zip)

Negsol, 2021-12-20:
Maybe this is enough of a fix. But maybe it has something to do with the constant polling.
I will test the new FW now.

Negsol, 2021-12-20:
nothing change when the controller is in the Xbox360 mode see log
[firmware211220f.txt](https://github.com/mist-devel/mist-firmware/files/7747876/firmware211220f.txt)

Negsol, 2021-12-20:
When I switch to HID under windows and plug it back into MiST - it will be recognized under an new VID PID:
VID 1001
PID 0002

Negsol, 2021-12-20:
Then not touch the mode button again will leave the controller in this state (also surviving cold restarts).

Negsol, 2021-12-20:
mode switching under the new firmware is still not possible

Negsol, 2021-12-20:
However I guess that no mode switch actually happens on MiST. That could be a miscommunication between the devices. The LED on the controller will not blink and do not flash as normal for a mode switch with this controller.

Negsol, 2021-12-20:
It appears that the controller simply remains in the mode it was last set to and also that the mode switching is disable as long as it is connected to MiST.

gyurco, 2021-12-20:
Ok, then I really don't know why it changed the mode previously, also I have no idea why XBOX mode doesn't work. These controllers have really strange firmwares.

Negsol, 2021-12-20:
Chinese special with a lot of copy&paste

Negsol, 2021-12-20:
Yes - but also be aware that this could be something affecting other controllers. But of course maybe the controller I have has the bug. No update there - I guess it was discontinued.... Anyhow big thanks for troubleshooting this!!!

gyurco, 2021-12-20:
It's hard to do tests without the actual controller - I got a strange one from a user, so I could debug it, it didn't work until at least the language ID descriptor was read. Then other controller broke when all strings was read. Brain-dead software run on these gamepads. BTW, I wonder if the real XBOX controller works at all.

Negsol, 2021-12-20:
I know - I remember when Till was asked about the implementation that he was happy with just supporting HID and the bare minimum to be as compatible as possible. Again I would suggest to at least consider making the new changes (xbox mode) an option to be disabled in the MiST.ini. I know we have parameters for mouse in there to ensure that compatibility is possible.

Negsol, 2021-12-20:
Anyway I write a tutorial for this controller so I remember later. Good also that we have this conversation here on github for others to see.

gyurco, 2021-12-20:
It wouldn't help disabling the XBOX driver - if the controller is in XBOX mode, then it won't work with the HID driver anyway.

Negsol, 2021-12-20:
true

Negsol, 2021-12-20:
Maybe we have to update the wiki (but who will read it). Also it is important the even when you change the mode to HID successfully the VID and the PID could have change. With my controller I have to edit the mist.ini to get the correct button mapping. Easy fix of course when you know the VID PID changed.

gyurco, 2021-12-20:
I wonder how many VID/PID pair a controller can have :)

Negsol, 2021-12-20:
lol

closed 2022-01-13 by Negsol