mist-devel/mist-firmware #86: core version not displayed anymore

jotego, 2022-01-18:
It looks like the core version as reported in the OSD string is not displayed anymore in the OSD menu. It used to be visible after the core (or ARC) name. This was a very useful feature for developing. I directly used the git commit as the version to keep track of things. Could this be included back, please?

gyurco, 2022-01-18:
Good catch. Actually it's displayed when 'V' string is parsed, e.g. you scroll to the end of the menu, or there's no more than 8 entries. But due to an optimization, reading the config string stops when the OSD is already filled. Currently it's possible to work around to put the 'V' string somewhere at the beginning of the config string. I'll try to do something to read it even it's at the end without sacrificing the optimization.

gyurco, 2022-01-18:
Fixed in https://github.com/mist-devel/mist-firmware/commit/a21bb80928587319022884547566a18c186a4025

closed 2022-01-18 by gyurco