mist-devel/mist-firmware #88: Core version hidden by firmware in OSD

jotego, 2022-03-19:
In firmware **220220**, the OSD only shows the name of the .arc file but it doesn't show the contents of the **V** line in the config string. This line is important because it lets the user identify which version of the core they are running. That helps debugging the cores. 

Could you restore the old behaviour and show the core version in the OSD?

Thank you

gyurco, 2022-03-19:
I thought it was already fixed:
https://github.com/mist-devel/mist-firmware/issues/86
I'll recheck.

gyurco, 2022-03-19:
Just loaded Golden Axe (great S16 finally released!), it displays "GOLDNAXE df574c6 - patreon.c". Is there any special you do to trigger the bug?

jotego, 2022-06-26:
I think this has been fixed for a while now.

closed 2022-06-26 by jotego