mist-devel/mist-firmware #91: Easier config string addressing

jotego, 2022-05-11:
The MiSTer guys are moving away from the alphabet-based config string and can use direct bit numbers now:

https://github.com/MiSTer-devel/Template_MiSTer/commit/79e54ac7a372d8d753bb7490d4f6e5e929fe8b5c

Just in case you consider this a valuable change.

gyurco, 2022-05-11:
I don't see the need for this until the status word length is 64 bits. If it'll be increased ever - then it can be done.

harbaum, 2022-05-12:
I also think that this is one of the things one would do as part of a bigger redesign. The entire menu engine is pretty inflexible and I'd redo it entirely for an updated device. But for the MIST I think the current solution is mostly sufficient.

jotego, 2022-05-12:
Thank you for reviewing it. if MiSTer stopped supporting the old naming scheme, it would make cores a bit harder to support across both platform. However, so far MiSTer is still supporting both namings. 

I'll close the issue.

closed 2022-05-12 by jotego

gyurco, 2022-05-12:
Throwing away support for the old scheme would be suicide, as it would break all old cores.

You're right about the redesign, for example I would support 'sparse' bits, like change bit 1 and 3, but leave bit 2 intact, add some more UI elements, etc.
If you have something in mind, just let me know.