mist-devel/mist-firmware #92: 8bitdo m30 2.4g not work.

xdemax, 2022-05-19:
hello! i have a fairly popular controller '8bitdo m30 2.4g' available, when connected via usb it works in xbox compatible mode (x-input), unfortunately mist does not detect it.

gyurco, 2022-05-19:
Need firmware logs to see what happens.
BTW, the XBOX driver is just partial, as I don't have too much XBOX compatible controllers to test.

xdemax, 2022-05-19:
ok! i understand and upload log file.
[log.txt](https://github.com/mist-devel/mist-firmware/files/8730451/log.txt)

xdemax, 2022-05-19:
that's right, the first turn on with a connected gamepad - 'joy 1 -' appears in the menu, and after rebooting the button disappears.

gyurco, 2022-05-19:
And did you reboot intentionally? Or it just happened by itself?

xdemax, 2022-05-19:
forced by button.

gyurco, 2022-05-19:
Yeah, that's the XBOX Controller VID/PID. Seems this controller also tries to be 'intelligent' and detects the host via its USB initialization pattern, and it switches to HID mode with 057E:2009 VID/PID after it detects something else (not Windows I guess). And why doesn't work in HID mode? That's another question...
Unfortunately it'll be very hard to debug this without having the actual device to experiment with.

xdemax, 2022-05-19:
according to the manual -- 'On USB connection, it works with Switch, Windows 10 and Raspberry Pi', 057E:2009 corresponds to joy-con nintendo switch.

gyurco, 2022-05-19:
The HID parsing aborts:
```
HIDP: PHYSICAL_MAXIMUM(0)
HIDP: unexpected main item 0
```
There must be some unusual data in the report descriptor  or an there's an issue with the report descriptor parser. I'll investigate. Maybe it would be easier to make it work in HID mode than find out how to stay in XBOX mode.

xdemax, 2022-05-19:
unfortunately this guy doesn't know how to be hid.
took another log, with the keyboard, mouse and gamepad connected.
[log2.txt](https://github.com/mist-devel/mist-firmware/files/8733667/log2.txt)

gyurco, 2022-05-19:
There are 3 data dumps from 3 HID initialization attempts, and all of them differ. Strange device. Maybe under-powered? The report descriptor dumps in the first log.txt are definitely invalid (and the two dumps even not the same), as I tried to load it into the official HID Descriptor Tool:
https://www.usb.org/document-library/hid-descriptor-tool

gyurco, 2022-05-19:
The first HID descriptor dump in log.txt from offset 0f to 3f is all zero, in the second dump this area is almost all zero. Usually this data should be hardcoded in the firmware, thus it should be identical in all queries. Looks like a hardware problem for me. Maybe you can try a different power supply? Or a powered HUB?

xdemax, 2022-05-19:
this log is obtained by connecting the gamepad after loading the core
[log3.txt](https://github.com/mist-devel/mist-firmware/files/8733908/log3.txt)
.

gyurco, 2022-05-19:
Another version of the report descriptor :)
But at least it's fully parsed, unfortunately at the end:
"HIDP:   - unusable report 130"
I wonder what would be the correct descriptor. Can you dump it somehow?
I found this tool for Windows:
https://github.com/todbot/win-hid-dump
But maybe it'll be in XBOX mode if you connect it to a Windows machine.

If you have a Linux machine, you can use usbhid-dump:
https://www.systutorials.com/docs/linux/man/8-usbhid-dump/

xdemax, 2022-05-19:
yep, on windows is work perfectly, but dump with error:

0000:0000: Microsoft - HID Button over Interrupt Driver
PATH:\\?\hid#acpi0011#3&30be55&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  05  01  09  0C  A1  01  85  01  09  0C  A1  02  09  C6  15  00 
  25  01  35  00  45  01  65  00  55  00  75  01  95  01  81  02 
  95  07  81  03  09  CB  95  01  B1  02  95  07  B1  03  C1  00 
  C1  00

gyurco, 2022-05-19:
Are you sure this was the 8bitdo device?
0000:0000: Microsoft - HID Button over Interrupt Driver

Isn't it possible to update the firmware of this gamepad? Looks very strangely behaving.

xdemax, 2022-05-19:
yes. when launching the application with a connected gamepad, the error 'unhandled exception -- failed to get info.'

xdemax, 2022-05-19:
I don't think it's worth wasting time on this.

jotego, 2024-04-15:
This is a controller I use too. I have also noticed this on/off behavior. It looks from the issue log that this cannot be fixed, at least not easily.

gyurco, 2024-09-25:
The latest firmware has some fixes related to this by UzixLS. Maybe you can try it.

jotego, 2024-09-29:
Thank you for the headsup. I upgraded the firmware and tested it but it made no difference :-(

jaz119, 2025-12-26:
I have M30, works well with MIST.