mist-devel/mist-firmware #95: Access SD card via MiST's USB interface

jotego, 2022-08-15:
This is a functionality that I have seen in the [Analogue Pocket](https://www.analogue.co/developer/docs/getting-started#usb-link-mode). It lets you see the contents of the SD card without pulling it out, when you connect the device to a computer via USB. It is slow, but it is faster than moving the card around.

I'm just tossing the idea in case you find it interesting for the MiST. MiSTer doesn't need this as the SD card can be accessed via ethernet.

gyurco, 2022-08-15:
It was implemented some month ago:
https://www.atari-forum.com/viewtopic.php?t=41183
Not very fast, but usable.

Just add
```
[menu]
usb_storage=1
```
to mist.ini
And you'll have usb storage when you're in the menu core.

jotego, 2022-08-16:
Awesome! I should check the forum more often.

I've just ordered the male-male USB cable to check it out, from the myriad USB cables I have, that type was missing.

Thanks a lot!

closed 2022-08-16 by jotego

gyurco, 2022-08-16:
You don't need a male-male cable, the device mode is available only at the micro USB connector (where you normally power the MiST).

jotego, 2022-08-16:
Blast! Thanks for telling me so quickly. I just canceled the order.

jotego, 2022-08-16:
Is it in the main firmware now or only in that beta you linked to?

gyurco, 2022-08-16:
No, it's part of the release in mist-binaries for a while.

jotego, 2022-08-16:
Thanks. Does it work while any core is on or only with the menu core?

gyurco, 2022-08-16:
It should work in any core, but when the MCU is busy for a long time, like ROM loading or other disk access, the host can detect the USB connection as dead, and that's not good for the health of the filesystem.