mist-devel/mist-firmware #99: The firmware doesn't load core.arc after a JTAG update

jotego, 2022-10-25:
I would have sworn that this worked a while ago... When you use the JTAG to load the core, the core.arc is not reloaded. So even if manually load the ROM, the DIP switches will be wrong. In some cases, wrong DIP switches will prevent the core from booting up (like CPS1).

Would it be possible to parse the core.arc after a JTAG load to get the DIP switches and OSD menu right? Re-sending the ROM file would be good too as the SDRAM often gets corrupted during the JTAG transfer.

Thank you

gyurco, 2022-10-26:
Core.arc is never checked, only when booting MiST normally, but the ROM file should be resent after a JTAG upload. Or you can try to press the ARM reset button while SW1 is ON - that will reboot ARM, but prevent reloading the core.{rbf, arc}.

Loading an ARC file _after_ the core is somewhat role reversal, also which ARC should be loaded for any cores? They usually have a long name, and there are no references to the ARC in the core itself (only vice-versa).

jotego, 2022-10-26:
It looks like it was my imagination then, and it never worked that way.

Ideally, if the ARM can detect that FPGA has been reprogrammed, it should load the core.arc file. I think it makes sense to assume that the core.arc game is the one the developer is working on as it is the one that gets loaded when the MiST is turned on.

It does not matter whether the RBF is already sent because the ARM can still get the config string and send the right default status bits and show the OSD correctly. It even can reset the core after setting the values in order for the retro system to get the status information (as some arcades only read the values upon reset)

I think the ROM isn't sent after a JTAG event. The core most of the time will hang up. Resetting it doesn't fix it but reloading the ROM manually does.

I still find myself pulling out the SD card as the default work method. Using the USB interface from the MiST to the PC is so slow that it is impractical (a real pity) and not having the right default DIP switches and missing the right OSD menu is often a problem too.

Is it very hard to detect the JTAG event and load the settings from core.arc?

gyurco, 2022-10-26:
Maybe it can be added to load debug.arc or devel.arc. Or a "corename.arc" where corename is the first item in the config string (as "normal" cores supply the core name). I would skip core.arc, as it's something for users.
I'll check the ROM loading.
Sorry about the USB speed, but that's what the hardware can do. You can speed it up a bit using a smaller SD Card for development with less files/directories, than the initial scanning from the OS side will consume a bit less time.

gyurco, 2022-10-26:
Just checked: the ROM is loaded after a JTAG programming, with the name inside the core (JTOUTRUN.ROM for example in your case). This can be circumvented by reading an appropriate ARC file beforehand (JTOUTRUN.ARC is in my mind in this case).

jotego, 2022-10-27:
I see! That's why the ROM didn't get loaded as I was using the name in the ARC. It must have never loaded for my games then but when I was working on small-sized games the chance of SDRAM corruption during downloading was smaller, so the game was able to restart after programming on some occasions.

Using an .ARC name based on the core name is a great idea. It makes possible to program different cores via JTAG and all of them will get the right file.

Thank you for the suggestion about the SD car size and number of files there. My SD card has many, many files so that must have been slowing things down.

gyurco, 2022-10-27:
Here's a compile firmware with the change.
[firmware.221027.zip](https://github.com/mist-devel/mist-firmware/files/9880056/firmware.221027.zip)

jotego, 2022-10-28:
This works like a charm. I have tested it with six different cores, making .ARC files with the name of the core and programming from the linux workstation. The OSD menu is right, the ROM gets loaded -with the name in the ARC, the games boot correctly...

Really, really useful!

Thanks a lot

jotego, 2022-10-28:
BTW, the file was a great -and unintended- birthday present for me :-P

closed 2022-10-28 by jotego

gyurco, 2022-10-28:
Cool! Happy birthday!

jotego, 2022-10-30:
I have added JTAG programming to JTFRAME's jtcore script. So the device can be programmed after compiling. Check this [commit](https://github.com/jotego/jtframe/commit/b7d8395619c7a25a150bdd2855fd8e2f60dfce7a).

The usage is: `jtcore outrun -p`

gyurco, 2023-04-18:
I just discovered that there's an even more convenient method to test various ARCs after JTAG programming:
Put SW1 into ON state to prevent reprogramming the FPGA, then go to the Fw&Core menu to change the core, and select and ARC: the ARC will loaded with the appropriate ROMs, but the FPGA will preserve the programmed state.