mist-devel/mist-firmware #104: GitHub Action for automatic test builds of the firmware

retrofun, 2023-03-28:
My [changes for a system status page](https://github.com/mist-devel/mist-firmware/commit/b685da627c984268f791c3a9046f29fb6e1b432d) introduced compile errors for older versions of gcc and were fixed [here](https://github.com/mist-devel/mist-firmware/commit/f20455657bd4286ce5dcb548510811d66e6a9e2a). Newer gcc versions accept new syntax constructs and thus hide such errors.

This can be prevented in the future with automatic test builds of the firmware.

I've built a Docker image with the (aged) build environment/cross compiler created by [install_arm-none-eabi-gcc.sh](https://github.com/mist-devel/mist-board/blob/master/tools/). A ready-built public Docker image to build the firmware is available at the GitHub Container Registry: [ghcr.io/retrofun/mist-firmware:builder](https://ghcr.io/retrofun/mist-firmware:builder).

Repository for the Dockerfile to create the Docker image is [docker-mist-firmware-builder](https://github.com/retrofun/docker-mist-firmware-builder).

Results of automatic test builds with GitHub Actions on push can be seen in my two test branches
* [github-actions-test](https://github.com/retrofun/mist-firmware/commits/github-actions-test): Failure, red cross @ 'Add test-build action' commit
* [github-actions-test-2](https://github.com/retrofun/mist-firmware/commits/github-actions-test-2): Success, green checkmark @ 'Add test-build action' commit

Click on red cross / green checkmark for details!

For mist-firmware repository it should be sufficient to cherry-pick the 'Add test-build action' commit (file `
.github/workflows/test-build.yaml`).

Or create an own Docker image, publish it as package for mist-devel, change the image URL in the `test-build.yaml` file and add it to the repository.

gyurco, 2023-04-02:
I'm not proficient in GitHub actions, but if it can be used with only a single YAML file, then why not?
But as there are only about 2 regular contributors, I don't know if it's a really big help :)

jotego, 2024-05-16:
It is true that, as @gyurco said, if the two regular contributors are already set, this may not be of much help. But the action file is also a way to document the tool chain to build the firmware. And the docker files will ease setup for the occasional user. 

@retrofun would you consider making a PR?