mist-devel/mist-firmware #105: Button mapping

jotego, 2023-04-30:
I was told yesterday how to remap buttons in MiST/SiDi. I had no idea it was possible. It's a great feature and most welcome but still troublesome. Let me explain:

If you start one of the Street Fighter games and try to remap the controls, it will ask you about start and select too early and the buttons the user presses then will get recognized by the core as actions (kick/punch). This is because the core is expecting coin/start to be the last two buttons. So the users ends up totally confused after the mapping.

I am not sure what the best route here is. Some options:

1. Make controller mapping per core, as MiSTer does, and add button names to ARC files, as MRA files do
2. Redesign cores so start/select always appear at the position the MiST menu expects, and other buttons after that

Probably both options imply breaking current mapping files for users. I don't mind going option 2 and recompiling cores if we agree to that but option 1 is the most user friendly overall.

What do you think?

gyurco, 2023-04-30:
I think about using the button names from the MRA, and display this instead of the standard ABCSelXY...etc during the manual configuration in the OSD.
I just not sure about the format of the configuration.
E.g. what does this mean (from 1941)?
```
    <buttons names="B0,B1,-,-,-,-,Start,Coin,Pause"
        default="A,B,R,L,Start"/>
```

A - joystick[4] -> B0
B - joystick[5] -> B1
Select, Start, X, Y - joystick[9:6] -> unused
L - joystick[10] -> Start
R - Joystick[11] -> Coin
L2 - joystick[12] -> Pause

And then what are the meanings of the defaults?

gyurco, 2023-04-30:
I found just a bit strange that without any custom mapping, joystick[7] is start, then it should be presented as joystick[10]?

gyurco, 2023-04-30:
Just worked a bit on the ARC file handling, added a BUTTONS= item. Attached a firmware with an 1941 ARC file if you want to see. I'm thinking about how useful would be the defaults, as without any mapping config, buttons like L and R are probably not where you expect them on most gamepads by default.
[fw-joymapping.zip](https://github.com/mist-devel/mist-firmware/files/11362478/fw-joymapping.zip)

jotego, 2023-05-02:
Thank you for looking into this

> ```
>     <buttons names="B0,B1,-,-,-,-,Start,Coin,Pause"
>         default="A,B,R,L,Start"/>
> ```
> And then what are the meanings of the defaults?

The dashes show buttons that a core support, but the game doesn't. In the case of JTCPS1, the core supports six buttons but some games only use 2 or 3. The dash indicates the system that it shouldn't ask for that button but should map the others to the right bit location.

Let me try the firmware

jotego, 2023-05-02:
It works perfectly! Thank you

Do you know if there is someone supporting the mra tool at the moment to add the new feature to ARC files?

gyurco, 2023-05-02:
> 
> The dashes show buttons that a core support, but the game doesn't. In the case of JTCPS1, the core supports six buttons but some games only use 2 or 3. The dash indicates the system that it shouldn't ask for that button but should map the others to the right bit location.

My question is that the number of buttons/dashes are always represent the expected position in the joystick[] vector?
E.g. Pause will be joystick[8] coming from user_io in every case?

> Do you know if there is someone supporting the mra tool at the moment to add the new feature to ARC files?

It was forked to mist-devel, maybe @squidrpi can add this change. Or I can do it at last resort, it doesn't seem to be hard.

gyurco, 2023-05-04:
Patch added to mra-tools:
https://github.com/mist-devel/mra-tools-c/commit/83ae3a1577b00510ffa78ad71192bec0459cd6e6

gyurco, 2023-05-05:
Released in firmware 230505