mist-devel/mist-firmware #106: Potential issue with 15kHz mode

slc79, 2023-09-04:
I've been testing the 15kHz mode for a while on a CRT TV and ever so often, it will flash with a black screen. I first thought it was down to the RGB signal not being switched properly but I have ruled that out. I currently have no other TVs to test on, but will attempt to get a Commodore 1940 set up to see if it may be sync related and that the TV is a bit more fuzzy about syncs.

Have anyone else experienced this? Using a MiST V1.2 with latest firmware and a VGA->SCART cable. Happens with all cores, so I don't believe it's down to any specific core. The MiST really shines on a 15kHz CRT. Wouldn't be able to tell it apart from the machines the cores implement in most cases.

gyurco, 2023-09-05:
Every core has its own way to create video in the FPGA, so it's surely not a firmware issue.

slc79, 2023-09-05:
OK, so I guess I should start by measuring actual frequencies from the various cores. The reason I was thinking it somehow lies with the firmware is because it happens to every core I try. Is there a chance that something could be stalling the part that controls the video output every now and then causing flicker, or may it be a matter of fine tuning the cores? If it only happened to one or two cores that would make sense, but since it happens to all of them it's easy to suspect something common.

Do you have any suggestions on things I could try, as I wouldn't entirely rule out something in my setup being at fault. :) Funny thing, though, as I was able to test on my C1940, and not a single flash (but the Spectrum core was unable to sync properly at 15kHz, so that one is surely core specific). 31kHz however worked flawlessly.

slc79, 2023-09-06:
OK, I will close this ticket now. It actually was a user error! Sorry! :)

The problem turned out to be my SCART switcher. I am not sure why it doesn't properly handle this as every other retro computer or console I attach to it works flawlessly. I started testing this on multiple other screens and none did have this flicker, except that particular Akai TV. So, the only thing different between different setups now was the SCART switch. I connected the RGB cable directly to the TV and issue went away.

In hindsight I realize I should have suspected the switch first, but it just seemed so unlikely since it never caused me any problems. PS! If I have any future questions, are there any other ways to reach any of the guys still involved with the project instead of creating a ticket?

To everyone involved, you make an awesome effort. I regret not having paid attention to this very simple and user friendly kit earlier. It suddenly gave me access to so many systems I currently don't own. My only other FPGA computer is the Ultimate 64 which opened up my eyes for the possibilities to keep platforms alive in a better way than most software emulators are capable of.

gyurco, 2023-09-07:
BTW, it's a surprise there are still new users for this old MiST :)

slc79, 2023-09-08:
It was a pure coincidence. I was offered various stuff for trade for a pristine/mint condition C64, and after talking to a previous MiST owner I was made aware that the Amiga core was no worse than the one on MiSTer which was the main selling point to me to go with the MiST. All the cores I've tested so far have been working pretty well too and the fact it's so easy to use and just ready in a box makes it an awesome device. I realize most people now do development for MiSTer but I've observed some work still being done for the MiST occasionally and I've found most cores I've been wanting, but nothing for CoCo or Dragon :( The AtariST core is also giving me some odd issues. Seems to be a resolution in there that nothing I own wants to sync properly to. The car selection screen won't give picture on any of my VGA capable screens and gives an improperly synced picture on my 15kHz displays. For someone wanting an easy FPGA kit for multiple platforms I would actually have no problem recommending a MiST over  MiSTer for the convenience alone, but in time I probably want the latter too :)

closed 2023-09-08 by slc79