mist-devel/mist-firmware #109: Allow long names in .rom files

jotego, 2023-11-02:
Because of the trimming done in .rom games, there are overlappings and some games cannot be played. It looks like long names are ok for arc files. Can we change that for .rom files too? Even if spaces are not allowed, let the file name be like 12.3 instead of 8.3

Two examples reported by an user:

> In the Outrun core "Out Run (deluxe sitdown) (Enhanced Edition v1.0.3)" and "Out Run (sitdown-upright, Rev B) (Enhanced Edition v1.1.0)" both create and use roms called "outrueha.rom"
In CPS2, "X-Men Children of the Atom (Asia 941219)" and "X-Men Children of the Atom (Euro 950105)" both create and use roms called "xmcotar1.rom"

gyurco, 2023-11-02:
Also the MRA tool got the change:
https://github.com/mist-devel/mra-tools-c/commit/dd89e254963d085c5030b96d3fbbf804ce0605ff

jotego, 2023-11-02:
That's great. I will update my workflow to remove the name trimming. This is going to cause some havoc...
Thank you

closed 2023-11-02 by jotego

gyurco, 2023-11-02:
Longer setnames are already used in some cases, like:
https://github.com/Gehstock/Mist_FPGA_Cores/blob/1a2492eb0adfcf9876bb8f24dbd00d960e23d3b8/Arcade_MiST/IremM72%20Hardware/meta/Air%20Duel%20(World%2C%20M72%20hardware).mra#L4