mist-devel/mist-firmware #111: Recover from a re-inserted SD card

jotego, 2024-02-09:
It would be nice if the firmware could recover from an extracted and re-inserted SD card without having to power cycle it. One use case for me is this:

1. Boot up a core
2. Play up to a critical point, where I save a .RAM file for debugging in the simulator
3. Extract SD card and copy to PC
4. Insert the card, reboot and start over, rather than continuing at the same point of the game

harbaum, 2024-02-09:
Resetting the IO controller using the MiSTs reset button should work without the need to power cycle, doesn't it?

jotego, 2024-02-10:
What I am trying to do is to keep the game at the same point, so I can make a save file at stage 1, then another at stage 2, then stage 3... without having to replay it each time.

harbaum, 2024-02-10:
So resetting the IO controller would do the trick as long as your core just ignores the fact that the io controller would send a reset signal and would try to upload ROMs.

If you add a signal to your core that ignores those things after you have saved the state, then your core would continue to run and can e.g. save another state like it did before.

gyurco, 2024-02-10:
I think the problem here is when the ARM is reset, it re-uploads the ROMs and sends a reset to the core.
But maybe it's also possible to make FatFS happy after a card remove-insert, too, there's not much data cached, so if the card is not written, it might be possible to continue.

gyurco, 2024-02-10:
Re-initing the filesystem after a card change is detected should be not that hard - MSDOS already did it 40 years ago :)
Interestingly, Linux systems were not that good in that, they couldn't even handle a RO CDROM change gracefully (they tend to crash if the disk was replaced without umounting).
FatFS usually calls the disk_status callback before every operation, probably that would be a good place to hook a reinit code.

gyurco, 2024-02-10:
Or probably only need to check periodically if the card is still inserted (calling MMC_CheckCard), as the logic to reinit is already there. The firmware simply doesn't know if it was removed.
mmc.c#L62https://github.com/mist-devel/mist-firmware/blob/38fca2e6e536031d09089e1b7736f0907df7db57/hw/AT91SAM/mmc.c#L62

jotego, 2024-02-10:
> Or probably only need to check periodically if the card is still inserted (calling MMC_CheckCard), as the logic to reinit is already there. The firmware simply doesn't know if it was removed. [...]

I like this one!

gyurco, 2024-02-10:
First step is done, now maybe FS reinit should be added, too (but if the card was only removed to copy something from it, then it's already working).

jotego, 2024-02-10:
> First step is done, now maybe FS reinit should be added, too (but if the card was only removed to copy something from it, then it's already working).

That's good enough for my use case. Thank you!

gyurco, 2024-02-11:
The newest release has this. Interestingly, core types MIST and MINIMIG did this thing already (they even unmounted all floppy drives when the card was removed).

closed 2024-02-12 by jotego