mist-devel/mist-firmware #115: core-specific OSD options (cfgstr) set in the ini file

jotego, 2024-05-16:
This is the MiST version of my [MiSTer issue](https://github.com/MiSTer-devel/Main_MiSTer/issues/901). We had touched on a related topic recently but I could not find the issue in mist-devel. We were discussing about using the *core_mod* information but it was limited to one byte. This topic is closely related:

Although each arcade core is a different RBF file, many cores come from the same author and hence follow some common settings and OSD arrangements. The 57 RBF files I am maintaining have a consistent OSD interface and status word. But other authors, including those publishing in the mister-devel area, are also managing settings uniformly.

It would be convenient to provide some way for the user to enter common information for cores in the ini file. A good use case is the CRT adjustments. Even today, CRT users are about 50% of MiSTer users according to polls I run periodically. If you have a CRT that shifts the image to the left, you currently need to adjust each core manually and save settings. It would be convenient if users could do something like this in the ini file:

```
[jt*]
CRT H offset: +8
```
In this case the CRT H offset string comes directly from the cfgstr provided by the core. The core name uses [glob patterns](https://en.wikipedia.org/wiki/Glob_(programming)).

But I do not care so much about these solution proposals. My point is that letting the user set common settings would be very convenient. Is this already supported? If not, may I please request support for this?

gyurco, 2024-05-16:
I rather think about adding a 'G'lobal specifier to the cfg string, and options marked with them are saved to a global .ini file instead. Like 'OxyG,CRT H offset,1,2,3,4'. Need to think about the key used in the .ini file.

gyurco, 2024-05-16:
What about this?
Have a 'Gsection;' string, which you can choose a section in a global .ini file. Then O..G[0-9], which can choose a global option in the appropriate section, like:
```
Gjt;
O34G0;CRT H offset,1,2,3,4;
O56G1;CRT V offset,1,2,3,4;
```

And the global ini could have:
```
[jt]
S0=2; CRT H offset
S1=3; CRT V offset
```
A the end there can be 10 global options/'namespace'.
This way the ini parser doesn't have to be modified at all (for dynamic keys), so it's fairly simple to implement.

jotego, 2024-05-17:
If we go that route, the ini file is not user friendly and becomes more obscure because of the name used for the settings. And the cfgstr becomes more complicated too with a new section, rather than using glob syntax in the ini.

Sorgelig does not want to support this on MiSTer either. So it is ok. I'll drop the request. Thanks for considering it.

closed 2024-05-17 by jotego

gyurco, 2024-05-17:
One advantage of this method would be that the user don't need to be aware of the ini file at all, everything can be done in the OSD.