mist-devel/mist-firmware #124: mra tool cannot parse a file

jotego, 2024-11-08:
The mra tool is not parsing this file correctly. It generates an .arc file with incorrect locations for the switches. It also produces a core dump.

[Tehkan World Cup (set 1).zip](https://github.com/user-attachments/files/17675503/Tehkan.World.Cup.set.1.zip)

gyurco, 2024-11-08:
It works for me, generating the ARC file:
```
[ARC]
RBF=JTWC
MOD=0
NAME=TEHKANWC
DEFAULT=0xFFFFFF0000
CONF="P1,Switches"
CONF="P1OGH,Difficulty,Very Hard,Hard,Easy,Normal"
CONF="P1OI,Timer Speed,55/60,60/60"
CONF="P1OJ,Demo Sounds,Off,On"
CONF="P1OKM,Coin A,2/3,2/1,1/6,1/5,1/4,1/3,1/2,1/1"
CONF="P1ONP,Coin B,2/3,2/1,1/6,1/5,1/4,1/3,1/2,1/1"
CONF="P1OST,1P time,2:30,2:00,1:00,1:30"
CONF="P1OZ,Game Type,Credit In,Timer In"
BUTTONS="button 1,Start,Coin,Core credits"
```

Are you using the latest one?

jotego, 2024-11-09:
It generates the file, but the DIP switch assignment is wrong. To test it, hold both 1P and 2P while resetting (F3) JTWC. It will enter a test menu where you can see the effect of the DIP switches. Change them in the OSD and you'll see that they do not match correctly.
However, when running the core on MiSTer, DIP switches operate correctly right off the MRA file.

gyurco, 2024-11-09:
I think it matches the MRA (base is 16):
GH=16,17 -> 0,1 in MRA
I=18 -> 2 in MRA
.
.
ST=28,29 -> 12,13 in MRA
Z=35 -> 19 in MRA

All of these has a correct 16 offset.
Something must be in the core.

jotego, 2024-11-10:
Sorry, it was my fault all the time. JTFRAME_DIPBASE should have been less than 16. Thank you for checking it.

closed 2024-11-10 by jotego