mist-devel/mist-firmware #125: Please add IMG as an extension for ST hard disk image

darwinmac, 2025-03-16:
The title really says it all. It would also work to remove the filter completely and leave it to the user to load a valid hard disk image. 

I realize IMG is also the extension for an ST graphics image, but it is the extension that the Mac (and, I think, Windows) sees as a mountable disk image.

darwinmac, 2025-03-16:
If the IMG extension is something you do not want to do, either the DMG or ISO extensions appear to work as well. My use case is to easily move the SD card between the MiST and my Mac, copy files onto the image, and use the hard disk image on my MiST. I think the DMG extension would only work for the Mac while I think .ISO would work for both Mac and Windows. My personal preference would be to have both extensions supported in case Apple changes something in the future.

gyurco, 2025-03-16:
Let's add IMG, ISO would be even more confusing, as it's used for CD images.

darwinmac, 2025-03-16:
> Let's add IMG, ISO would be even more confusing, as it's used for CD images.

I was just trying to give you options. IMG appears to be mountable on both macOS and Windows 11 (probably earlier as well). I agree with you on ISO.

gyurco, 2025-03-17:
It's in git, it'll be included in the next fw release.

closed 2025-03-17 by gyurco