mist-devel/mist-firmware #126: [Feature Request] Add different core.rbf name for Poseidon board

Maverick-Shark, 2025-03-17:
Hi,

     SiDi128 has different menu core file (core.rbf) renamed as sidi128.rbf  then, could we do the same for Poseidon (ep4cgx)? Like psd4cgx.rbf or as you want.

Thanks and best regards,
Shark

gyurco, 2025-03-17:
SiDi128 uses a different firmware, but the Poseidon uses exactly the same as MiST. You can compile your own with any name you want.

gyurco, 2025-03-17:
...by using the FW_ID macro.

closed 2025-03-17 by gyurco

Maverick-Shark, 2025-03-17:
Hi,

Sorry, but I need help about the compilation process. I mean, I don't know how to use the FW_ID macro. Please, could you help me?

Thanks,
Shark

naeloob, 2025-03-28:
I think he wanted to say "Using DEFAULT_CORE_NAME Macro", as it seems to define the default rbf name the firmware loads at start.

gyurco, 2025-03-29:
Yepp, sorry, that's the correct macro name.
Look at Makefile.SAMV71 for an example usage.